Berny23/LD-ToyPad-Emulator #1: i did everything and even started the emulator on my ip address but doesn't connect to cemu or LD

Lego-Dimensions, 2021-08-08:
so i just completed your tutorial and got to
node index.js
and it made me a ip address on the web with the emulator
now when i start lego dimensions i get the "Please Connect the ToyPad to the Console" 
message i even placed Batman for example on the emulator and got this:

pi@raspberrypi:~ $ node LD-ToyPad-Emulator/
Server running on port 80
Placing character: 1
Character placed: 1
(node:1959) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
EVENT: Event { pad: '2', index: 1, dir: 0, uid: '04A31DD42B4980' }
events.js:170
      throw er; // Unhandled 'error' event
      ^

Error: ESHUTDOWN: cannot send after transport endpoint shutdown, write
Emitted 'error' event at:
    at errorOrDestroy (internal/streams/destroy.js:107:12)
    at onwriteError (_stream_writable.js:436:5)
    at onwrite (_stream_writable.js:461:5)
    at fs.write (internal/fs/streams.js:307:14)
    at FSReqCallback.wrapper [as oncomplete] (fs.js:525:5)

Berny23, 2021-08-08:
First a quick question: Did you use the correct micro USB port on your Pi Zero? There are 2 of them, but only one can send data.

This error happens when you place characters before the game recognises the toypad. Try the following steps:
1. Close cemu and stop the emulator (CTRL + C)
2. Reboot the Pi Zero with "sudo shutdown -r now"
3. Wait for Windows to recognise the Pi Zero as "LEGO READER 2.1", it should automatically install a driver (check the notifications)
4. If you heard the Windows sound that indicates a new USB device and the driver was installed, pkease start the server via ssh but don't do anything with it.
5. Start cemu and the game
6. Choose a savegame or start a new game and wait until it prompts you to place a character.
7. Proceed normally.

Lego-Dimensions, 2021-08-08:
does it work with raspberry pi 4 b 8gb ?

Lego-Dimensions, 2021-08-08:
please man help me i don't get any notifications neither the windows recognises the raspberry pi 
what can i do

Berny23, 2021-08-08:
Please read the Readme file. It says that you need a device with USB OTG support. Of all the Raspberry Pi's only the Zero W will work because it has WiFi and USB OTG.

Berny23, 2021-08-08:
The OTG is needed because the Raspberry has to disguise itself as the real ToyPad.

Lego-Dimensions, 2021-08-08:
we don't have that in our country what can i do?

Berny23, 2021-08-08:
A last option to try would be manually running the USB script: "sudo bash usb_setup_script.sh" Just the output here.

Lego-Dimensions, 2021-08-08:
okay if you will help me i will pay you if it's necessary so you say there is a last option?

mkdir: cannot create directory ‘g1’: File exists
mkdir: cannot create directory ‘strings/0x409’: File exists
mkdir: cannot create directory ‘functions/hid.g0’: File exists
usb_setup_script.sh: line 17: echo: write error: Device or resource busy
usb_setup_script.sh: line 18: echo: write error: Device or resource busy
mkdir: cannot create directory ‘configs/c.1’: File exists
mkdir: cannot create directory ‘configs/c.1/strings/0x409’: File exists
ln: failed to create symbolic link 'configs/c.1/hid.g0': File exists
usb_setup_script.sh: line 31: echo: write error: Device or resource busy

Lego-Dimensions, 2021-08-08:
can u do a step by step guide on what can i do?

Lego-Dimensions, 2021-08-08:
i will be MASSIVELLY thankfull to you <3

Berny23, 2021-08-08:
It looks good, my output is axactly the same. Please check if your Windows has the latest updates and try to use a different cable. It won't work with adapters.

Lego-Dimensions, 2021-08-08:
what type of cable?

Berny23, 2021-08-08:
USB A to micro USB A. Please read the Readme file of this repository!

Lego-Dimensions, 2021-08-08:
I DID IT!

Lego-Dimensions, 2021-08-08:
but do you know how to place the bat mobile in the last part of the first level?

Lego-Dimensions, 2021-08-08:
[image]

Lego-Dimensions, 2021-08-08:
after this i will close this succesfully

Lego-Dimensions, 2021-08-08:
and it says an update is required
too

Berny23, 2021-08-08:
Awesome that it works, seems like you don't actually need OTG support!

This is a bit of a problem because the emulator can't write data yet. You have to restart the game and server and load your save. Then place Wyldstyle, Gandalf and then place a character like Bat Girl that also has the grappling ability. (Not Batman!)
After that, place the normal Batmobile, build the portal and drive onto the thing on the right side. As far as I know, this is the only time in the game where these steps are required. Later you can just press B when you're told to place a vehicle.

Lego-Dimensions, 2021-08-08:
thanks <3 see you man i realistically hope and truly hope you will be lucky, healthy and happy this month <333
again thanks so much you just made my dream come TRUE !!!!!!!

Berny23, 2021-08-08:
Thank you too for using my software :)
I'm going to close this now.

closed 2021-08-08 by Berny23

ignacioat2, 2022-08-09:
> First a quick question: Did you use the correct micro USB port on your Pi Zero? There are 2 of them, but only one can send data.
>
> This error happens when you place characters before the game [...]

i have the same problem but i have a raspberry pi zero wh and t
he lego reader  2.1 does not install.