Berny23/LD-ToyPad-Emulator #2: Can't find files

jhunter999, 2021-08-14:
pi@raspberrypi:~ $  sudo cp usb_setup_script.sh /usr/local/bin/toypad_usb_setup.sh
cp: cannot stat 'usb_setup_script.sh': No such file or directory
pi@raspberrypi:~ $  sudo chmod +x /usr/local/bin/toypad_usb_setup.sh
chmod: cannot access '/usr/local/bin/toypad_usb_setup.sh': No such file or directory

This is what pops up one trying to do step 3, don't have any knowledge into code so if you can tell me why it's doing this it would be a big help!

Navex, 2021-08-15:
You have to clone the repository first before you execute the command to copy `usb_setup_script.sh` the read me is a bit confusing. Just run the commands in this order

```
    sudo apt install -y git
    git clone https://github.com/Berny23/LD-ToyPad-Emulator.git
    cd LD-ToyPad-Emulator
    
    sudo cp usb_setup_script.sh /usr/local/bin/toypad_usb_setup.sh
```

Berny23, 2021-08-15:
Thanks for the help @Navex!
I think I messed up the command order a bit. Fixing it soon.

closed 2021-08-15 by Berny23

Navex, 2021-08-15:
I made a pull request that fixes it

Berny23, 2021-08-15:
@Navex Thanks. I didn't see it before closing this issue.

commit 9fffed38 (2021-08-15): fix #2