Berny23/LD-ToyPad-Emulator #4: Rpcs3 emulator does not work at all

Yoshi9800, 2021-08-15:
When use rpcs3 emulator say needs toy pad after run emulator even thier cemu works. 
[image]

Berny23, 2021-08-15:
Please see this post: https://forums.rpcs3.net/thread-201525-post-313832.html#pid313832
Maybe your issue can then be solved. I don't have RPCS3 installed currently, but it may be a problem with the emulator itself. If you have a real toy pad, please try to connect it and check if it works.

Yoshi9800, 2021-08-15:
Problem is outher guy useing rpcs3 0.09 build useing 0.1.7

Yoshi9800, 2021-08-15:
Also newest build has no portal bugs like post said

Yoshi9800, 2021-08-15:
@Berny23 my guy test newer build still bug out need look fixed for rpcs3 emulator because emulator toy pad does not work on Rpcs3 emulator