Berny23/LD-ToyPad-Emulator #9: How can I do this on a rooted android phone?

HamzehTamimi, 2021-09-18:
Hello, can you tell me how can I do this on a rooted android phone since I don't own a raspberry pi?

Berny23, 2021-10-03:
You may be able to do it by enabling the "usb_gadget" feature. It can be done with this app: https://f-droid.org/packages/net.tjado.usbgadget/

HamzehTamimi, 2021-10-04:
Can you tell me the steps to get it working using a rooted ... Thanks

Berny23, 2021-10-04:
I don't have a rooted phone available. You can look into the setup script and the code of the emulator to see what features need to be available/working.

Berny23, 2021-11-10:
It's now possible to run the emulator in a virtual machine on Windows, macOS and Linux!
I updated the guide and made a tutorial video here: https://www.youtube.com/watch?v=5PARAnrt1jU

closed 2021-11-20 by Berny23