Berny23/LD-ToyPad-Emulator #10: Confirm it works great on ps3

lapieandalpie, 2021-10-03:
We were about to mothball our ps3 when we found this project.  Got it running on a pi 4B and hooked to our old first generation ps3.

lapieandalpie, 2021-10-03:
The PS3 didn't provide enough usb power to boot the pi 4, so we had to hook an external power supply feeding 5v into the "GPIO" power pins and SSH remotely over the wan via laptop terminal. Externally powered, the game recognized the toypad right away.   About once every 40 minutes or so we lost the ability to remove characters from the pad. (app would show them removed but they wouldn't remove) and refreshing the app page just loaded a page with no characters visible.  The solution was to reboot the pi... although inelegant.  I wondered if there could be a "reboot server" button on the web interface and a way to set the pi to boot into toypad mode so we wouldn't have to ssh into the pi in order to boot/reboot.  That said... DUDE!!!! WHOAH!!!! How freakin epic is this project!!!!   Me and my son were giggling with glee the entire time we played, being able to access all the characters and levels we couldn't find/buy.   Thank you SO much!!! (sent a coffee your way)

Berny23, 2021-10-03:
**Thank you for the support and your infos about compatibility with the PS3!**

@cort1237 recently updated the layout and added the ability to locally save changes to toy tags, as well as many other improvements.
To update the emulator, just view the "Update" section in the guide.

closed 2021-11-06 by Berny23

the-onechiel, 2023-08-25:
Do you need the actual game disc or would it also work with an ISO file on a jailbroken ps3 slim console, because it doesn't work for me using the latter.

Ravandragp, 2023-12-17:
hello i want to try it on my ps3 too but what do i have to do first if i want to play on ps3?

the-onechiel, 2023-12-17:
For me it works yeah, I thought it didn't at first but you need to have one
port on raspberry for powering and the other connected tot the ps3

cort1237, 2023-12-17:
@Ravandragp 
> hello i want to try it on my ps3 too but what do i have to do first if i want to play on ps3?

You will need the game itself and then follow the Single-Board Computer (Raspberry Pi) method in the readme.

PeterDonald313, 2025-04-03:
I would also like to confirm that it works on ps3 as I am using a Raspberry Pi Zero W (https://www.sgeducation.ie/raspberry-pi-zero-w?src=raspberrypi) and it works extremely well! The vehicle upgrading works, it works on my brother's ps4 and it doesnt really crash ever! I would like to send you a coffee soon, just not at the moment.
Thanks very much for this!