Berny23/LD-ToyPad-Emulator #19: Raspberry Pi B3 -bash: cd: LD-ToyPad-Emulator: No such file or directory

manbat0213, 2021-11-04:
I'm probably just missing something, but when I run `cd LD-ToyPad-Emulator` it returns `-bash: cd: LD-ToyPad-Emulator: No such file or directory`  When continuing on and trying npm install, I get:

```
npmWARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'

npm notice created a lockfile as package-lock.json. You should commit this file.

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'

npm WARN pi No description

npm WARN pi No repository field.

npm WARN pi No README data

npm WARN pi No license f
```

I'm probably doing something wrong, but it seems to be a great program, and I would love to try and perhaps recommend it.

Berny23, 2021-11-04:
Seems like you didn't clone the repository yet. You skipped this line in the guide:
`git clone https://github.com/Berny23/LD-ToyPad-Emulator.git`

manbat0213, 2021-11-04:
So I input the whole thing this time (I was copying and pasting and didn't realize that it cut off halfway through step 3) but now I'm getting this: 
```
Computing checksum with sha256sum
Checksums matched!
Now using node v11.15.0 (npm v6.7.0)
Creating default alias: default -> 11 (-> v11.15.0)
pi@raspberrypi:~ $ sudo setcap cap_net_bind_service=+ep `readlink -f \`which node\``
pi@raspberrypi:~ $
pi@raspberrypi:~ $ cd LD-ToyPad-Emulator
pi@raspberrypi:~/LD-ToyPad-Emulator $ npm install
npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but npm-shrinkwrap.json was generated for lockfileVersion@0. I'll try to do my best with it!
npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)

> usb@1.8.0 install /home/pi/LD-ToyPad-Emulator/node_modules/usb
> node-gyp-build

gyp: Undefined variable napi_build_version in binding.gyp while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/home/pi/.nvm/versions/node/v11.15.0/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:345:16)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 5.10.17-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v11.15.0/bin/node" "/home/pi/.nvm/versions/node/v11.15.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/LD-ToyPad-Emulator/node_modules/usb
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN ws@8.2.3 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@8.2.3 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.8.0 install: `node-gyp-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.8.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-11-04T19_25_03_058Z-debug.log
```

Berny23, 2021-11-04:
Type `nvm install node` and then try again.

manbat0213, 2021-11-04:
```
 npm install
npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but npm-shrinkwrap.json was generated for lockfileVersion@2. I'll try to do my best with it!
npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm WARN deprecated node-pre-gyp@0.6.39: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm WARN deprecated hawk@3.1.3: This module moved to @hapi/hawk. Please make sure to switch over as this distribution is no longer supported and may contain bugs and critical security issues.
npm WARN deprecated request@2.81.0: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated tar@2.2.2: This version of tar is no longer supported, and will not receive security updates. Please upgrade asap.
npm WARN deprecated boom@2.10.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated sntp@1.0.9: This module moved to @hapi/sntp. Please make sure to switch over as this distribution is no longer supported and may contain bugs and critical security issues.
npm WARN deprecated cryptiles@2.0.5: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated hoek@2.16.3: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated har-validator@4.2.1: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.

> usb@1.8.0 install /home/pi/LD-ToyPad-Emulator/node_modules/usb
> node-gyp-build

gyp: Undefined variable napi_build_version in binding.gyp while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/home/pi/.nvm/versions/node/v11.15.0/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:345:16)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 5.10.17-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v11.15.0/bin/node" "/home/pi/.nvm/versions/node/v11.15.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/LD-ToyPad-Emulator/node_modules/usb
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN ws@8.2.3 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@8.2.3 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.8.0 install: `node-gyp-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.8.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-11-04T21_19_08_718Z-debug.log
```

cort1237, 2021-11-05:
@Berny23 I've given it a look and it seems like maybe some dependency updated and it broke the build.
I have attempted many fresh installs and all come up with an error. The first error showing up seems to be at node-usb's install, I tried downgrading that an earlier version and had some success but then serialport would show similar errors. 

I'm still looking into it, but a lot of this is out of my knowledge. Perhaps you could give it a look when you have the chance
[Regular Log](https://github.com/Berny23/LD-ToyPad-Emulator/files/7480044/2021-11-05T01_07_34_569Z-debug.log)
[When forcing USB 1.7.2 instead of 1.8.0](https://github.com/Berny23/LD-ToyPad-Emulator/files/7480046/2021-11-04T23_58_16_113Z-debug.log)

Berny23, 2021-11-05:
@cort1237  @manbat0213 
I did it! It's finally working again.

[image]

I updated the dependencies of my fork of node-ld (https://github.com/Berny23/node-ld). I also updated the package.json file of the emulator in this repository to use my fork instead of the 4 years old version from the npm repo.
This had to be done, because a few dependencies don't have a build for the Raspberry Pi architecture, so they have to be compiled locally, which always failed for some reason. Using the latest alpha version of node-usb, serialport and node-hid fixed this problem. This also fixed the vulnerabilities.

So, to get it working, just run the following commands:
```
cd LD-ToyPad-Emulator
git reset --hard
rm -rf node_modules
rm package-lock.json
rm npm-shrinkwrap.json
git pull
npm install
node index.js
```

cort1237, 2021-11-05:
Cool just tested it with a fresh install and it successfully installed. Thanks for figuring that out.

manbat0213, 2021-11-05:
So I have the web interface up and running.  However, when I try to load the game, I get "Please connect the ToyPad" on my PS5.  It may be because it's not compatible, so I'm going to try it on my PS4, but it will be a while before I can.  Here is what the Pi outputs when I try to add a character to the toypad:
```
EVENT: Event { pad: '1', index: '2', dir: 0, uid: '043A774BD1CA80' }
events.js:170
      throw er; // Unhandled 'error' event
      ^

Error: ESHUTDOWN: cannot send after transport endpoint shutdown, write
Emitted 'error' event at:
    at errorOrDestroy (internal/streams/destroy.js:107:12)
    at onwriteError (_stream_writable.js:436:5)
    at onwrite (_stream_writable.js:461:5)
    at fs.write (internal/fs/streams.js:307:14)
    at FSReqCallback.wrapper [as oncomplete] (fs.js:525:5)
```

cort1237, 2021-11-05:
Yeah I don't think we've had anyone try on PS5. Let us know what happens on PS4. The error being thrown occurs when the emulator tries to add a character to the game before a connection is established with the game.

manbat0213, 2021-11-06:
It's doing the same thing on the 4 as on the 5.

Berny23, 2021-11-06:
This guy got it working on PS4, so it must be possible: https://www.reddit.com/r/Legodimensions/comments/pb32zg/toypad_emulator_of_lego_dimensions/hamfj29/?context=3

manbat0213, 2021-11-06:
I'm thinking I might need an RP zero or a 4 for it to work right.  Has anyone gotten it to work with a 3B?

cort1237, 2021-11-06:
I did some reading and learned the that raspberry pi 3 is not capable of usb_gadget which is what allows the function to act as a usb peripheral. It's missing pins in the USB port that allows it to get commands through USB. Sorry for the confusion I'm gonna add this to the readme The best bet would be to get a Pi Zero W.

Berny23, 2021-11-06:
Oh, seems like I forgot to mention it in the prerequisites. The support of the usb_gadget API in the Linux kernel (and hardware) is necessary to make the "usb_setup_script.sh" work. This script lets the Raspberry Pi act as a real toy pad for any consoles connected.
Any single board computer that has an OTG USB port should be capable of this.

closed 2021-11-06 by Berny23