Berny23/LD-ToyPad-Emulator #22: Can't get console to read emulator

Luigimeansme, 2021-11-18:
I finally got my Pi in and was futzing around with it for a few hours, finally managing to get the emulator installed. Going to the IP address in my browser shows the emulator just fine, but I can't get either of my consoles (PS5 or Wii U) to read the emulator when I plug it in. What can I do to fix this?

cort1237, 2021-11-19:
What model Pi is it? 

Make sure you're using the proper USB port. For example on the Pi Zero W, looking at it from the side, the port on the right is solely for power and the one on the left supports OTG connections which the game uses. If all else fails try a different USB cable.

Luigimeansme, 2021-11-19:
> What model Pi is it?
> 
> Make sure you're using the proper USB port. For example on the Pi Zero W, looking at it from the side, the port on the right is solely for power and the one on the left supports OTG connections which the game uses. If all else fails try a different USB cable.

I'm using the recommended Raspberry Pi Zero W with the USB OTG cable that came with the starter kit plugged into the left port and using a USB-A to USB-A cable to connect it to my PS5 and it's not working.

cort1237, 2021-11-19:
> using a USB-A to USB-A cable to connect it to my PS5

So you're using a USB A dongle to as an extension to the PS5? Have you tried it without that?

Luigimeansme, 2021-11-19:
I unfortunately do not have a Micro USB data cable and I don't want to buy one at the risk of this not working

Luigimeansme, 2021-11-19:
I have also tried using a USB-A to USB-C cord on the PS5's USB-C slot and it didn't work there either

Luigimeansme, 2021-11-19:
Is it because I didn't plug it into my PC originally so it never set up OTG?

cort1237, 2021-11-19:
No it wouldn't be that. OTG just represents the types connections present in the USB port. It's a physical thing. It's hard to say what would be causing this as we're routing the data through the another party with the extension cable. 

Here's what I see:
- It could be the extension cable messing up the OTG connection and not allowing the game to properly communicate with the Pi.
- It could be something wrong with the Pi.
- It could be something wrong with the install.

So I recommend:
- First you said you had a Wii U do tests on that first bc we have confirmed working the emulator with Wii U.
- Reinstall the emulator onto the Pi
- If that doesn't work, get a USB A to micro USB cable and try it with that.
- If that doesn't work it may be something bigger wrong, so follow up on here.

Berny23, 2021-11-19:
@Luigimeansme Please don't use an OTG cable, it doesn't work like that. You have to connect a normal micro usb cable (like from your phone) to the Raspberry Pi Zero W and then to the console/PC without any adapter.

I know this, because I tried it myself. What you need is just one of these (for example): https://m.media-amazon.com/images/I/61iKClEm0tL.jpg

Luigimeansme, 2021-11-19:
@Berny23 That was it. I got a data micro usb cable off Amazon and tested it out, and it works great! Can now confirm this works on real PS5

cort1237, 2021-11-19:
Awesome enjoy the game

closed 2021-11-19 by cort1237