Berny23/LD-ToyPad-Emulator #23: Minor issue: some vehicles mislabeled

Luigimeansme, 2021-11-19:
Looking at the list of vehicles, I'm seeing that Cragger's ship is out of order in organization. It's showing that Swamp Skimmer (the default) is one of the rebuilds and Cragger's Fireship (a rebuild) is the default. Some other typos include:
"Cloud Cukko Car" instead of "Cloud Cuckoo Car" 
"Gyro Sphere" instead of "Gyrosphere"
"BIONIC STEED" in all caps
"The Jokers Chopper" instead of "The Joker's Chopper"
These are incredibly minor, just thought I'd point it out anyway

Berny23, 2021-11-19:
Thanks for pointing that out.
I will fix it soon in both this repository and my fork of node-ld (the origin of the list).

closed 2021-11-19 by Berny23

Luigimeansme, 2021-11-21:
@Berny23 Names have all been updated, but the app still claims Cragger's Swamp Skimmer is the second form, not the first form, even after deleting and creating a new one

Berny23, 2021-11-21:
Then, the IDs got mixed up.

Does the game spawn Cragger's Fireship if Swamp Skimmer?

Luigimeansme, 2021-11-21:
> Then, the IDs got mixed up.
> 
> Does the game spawn Cragger's Fireship if you select Swamp Skimmer?

Yes it does.

Berny23, 2021-11-21:
Please try it with the new update.

Luigimeansme, 2021-11-21:
> Please try it with the new update.

Almost. Swamp Skimmer now summons in Fireship, Fireship now summons in Sub, Sub now summons in Skimmer

Berny23, 2021-11-21:
Ok, now it should work.

Luigimeansme, 2021-11-21:
That fixed it, thank you! If I notice anything else wrong with vehicle IDs, I'll post it here

Luigimeansme, 2021-12-12:
@Berny23 found another vehicle issue. When upgrading Lloyd's Golden Dragon to third form and saving it, the emulator just marks the square as "Unknown." It summons in just fine, but it looks like the emulator doesn't have a label assigned to the Mega Flight Dragon's ID.

Berny23, 2021-12-13:
@Luigimeansme Please tell me what ID number the "Unknown" vehicle has. You can find it by right-clicking the square -> Inspect (element) -> look for "data-id=" in the selected line (not data-uid).

Luigimeansme, 2021-12-13:
> @Luigimeansme Please tell me what ID number the "Unknown" vehicle has. You can find it by right-clicking the square -> Inspect (element) -> look for "data-id=" in the selected line (not data-uid).

If I am looking in the right area, it says the ID is 1144. I also went ahead and tested creating a separate Mega Flight Dragon square and summoning it in, and it just gave the "An update is required to use this content" message. The ID for this one says 1134, so it looks like it's just a small typo

Luigimeansme, 2021-12-16:
@Berny23 Realized I forgot to ping you again and it's a closed issue so you might not've seen it. My previous post has the IDs needed

Berny23, 2021-12-16:
@Luigimeansme Ok, for some reason the third form of the dragon has the ID 1144, but the other two are 1132 and 1133.
I switched the labels, please update the emulator.

Luigimeansme, 2021-12-16:
@Berny23 That fixed it, thanks! I'm working on a 100% run rn so if I notice any other busted IDs, I'll let you know

Berny23, 2021-12-16:
Nice! 
The whole reason why I created this emulator, was because I also want to 100% it one day.

Luigimeansme, 2021-12-17:
@Berny23 Do you possibly have a Discord account? It would make things easier to point out rather than constantly mucking up the issues page here.

Luigimeansme, 2022-01-16:
@Berny23 Found another one. KITT Jet and Goliath Armored Semi are backwards, upgrading to tier 3 will label it as Goliath Armored Semi and upgrading to tier 2 will label it as KITT Jet

Luigimeansme, 2022-07-24:
@Berny23 I found one more mislabled vehicle. The DeLorean builds 2 and 3 are swapped, so build 2 (Electric Time Machine) is labeled as the Ultra Time Machine, and build 3 (Ultra Time Machine) is labeled as the Electric Time Machine

Berny23, 2022-08-04:
Fixed it. https://github.com/Berny23/LD-ToyPad-Emulator/commit/f545175086fec7967f41c67ddf02b12221d5cfe2

Luigimeansme, 2022-10-28:
@Berny23 Found another one: Sensei Wu's Flying White Dragon is mislabeled as "Flight White Dragon." Sorry if this is getting annoying!

commit 4ddb9e99 (2021-11-19): fix #23
commit 96df7f31 (2021-11-21): should fix #23
PR #87 (2022-10-30): Bug tag name and abilities
Tried to fix naming and abilities.
Partial fix #83 
Fix #23 

@Luigimeansme we need good source of abilities to review all tags