Berny23/LD-ToyPad-Emulator #25: Trouble installing on Pi 400 - WARN install No prebuilt binaries found

kapone3047, 2021-11-20:
I'm trying to install this on a Raspberry Pi 400 (which is capable of USB gadget mode), but am having trouble on the last install step.

After trying 'npm install'

```
> @serialport/bindings@9.2.7 install /home/pi/LD-ToyPad-Emulator/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=11.15.0 runtime=node arch=arm libc= platform=linux)
gyp ERR! configure error
gyp ERR! stack Error: Command failed: /usr/bin/python -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack                       ^
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:299:12)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:999:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:266:5)
gyp ERR! System Linux 5.10.63-v7l+
gyp ERR! command "/home/pi/.nvm/versions/node/v11.15.0/bin/node" "/home/pi/.nvm/versions/node/v11.15.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/LD-ToyPad-Emulator/node_modules/@serialport/bindings
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN ws@8.2.3 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@8.2.3 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.2.7 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.2.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-11-20T05_51_58_466Z-debug.log
```

I'm assuming this means I need to compile my own build, but don't know what I need to do (I'm not a dev, just a geek with some passing familiarity with Linux). Can anyone point me towards what I need to do?

Berny23, 2021-11-20:
See issue #21 for a solution.
I just didn't have time to add it to the guide.

closed 2021-11-20 by Berny23

kapone3047, 2021-11-21:
I'm still having trouble.

Started again with a clean Raspi image. Went through the installation steps but stopped before `npm install` and ran the following to ensure the right version of npm was used by node-gyp

```
npm install --global node-gyp@latest
npm config set node_gyp $(npm prefix -g)/lib/node_modules/node-gyp/bin/node-gyp.js
```

Then went to do `npm install` and got the following:

```
> @serialport/bindings@9.2.7 install /home/pi/LD-ToyPad-Emulator/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=11.15.0 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/home/pi/LD-ToyPad-Emulator/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:78:51: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
   78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
      |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:135:53: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  135 |   uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:175:52: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  175 |   uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:215:52: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  215 |   uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:271:50: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  271 |   uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Get(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:316:50: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  316 |   uv_queue_work(uv_default_loop(), req, EIO_Get, (uv_after_work_cb)EIO_AfterGet);
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE GetBaudRate(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:366:58: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  366 |   uv_queue_work(uv_default_loop(), req, EIO_GetBaudRate, (uv_after_work_cb)EIO_AfterGetBaudRate);
      |                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Drain(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:412:52: warning: cast between incompatible function types from ‘void (*)(uv_work_t* ’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  412 |   uv_queue_work(uv_default_loop(), req, EIO_Drain, (uv_after_work_cb)EIO_AfterDrain);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: At global scope:
../src/serialport.cpp:433:28: warning: unnecessary parentheses in declaration of ‘ToParityEnum’ [-Wparentheses]
  433 | SerialPortParity NAN_INLINE(ToParityEnum(const v8::Local<v8::String>& v8str)) {
      |                            ^
../src/serialport.cpp:452:30: warning: unnecessary parentheses in declaration of ‘ToStopBitEnum’ [-Wparentheses]
  452 | SerialPortStopBits NAN_INLINE(ToStopBitEnum(double stopBits)) {
      |                              ^
In file included from ../../../nan/nan.h:58,
                 from ../src/./serialport.h:13,
[... cut ...]
npm WARN ws@8.2.3 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@8.2.3 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself.

added 164 packages from 243 contributors and audited 164 packages in 128.117s
found 0 vulnerabilities
```

As I said, I'm not a dev, some I might be missing something that's obvious to someone that uses npm and node frequently, but simply following the instructions linked to from issue #21 didn't work completely.

Berny23, 2021-11-21:
What do you mean?
The log you posted clearly shows that everything was installed successfully and that there are no vulnerabilities.

Just continue in the guide.

kapone3047, 2021-11-21:
Sorry, I saw the warnings about peer dependencies and type miscasting and assumed it failed.

kapone3047, 2021-11-22:
Thanks @Berny23 have the software running fine now. Just need to work out how to get the Pi 400 connected to the PS5 as a USB gadget. Haven't had any luck so far, but will take that question over to the discussions.

commit 2e183aac (2021-11-20): fix #21, fix #25