Berny23/LD-ToyPad-Emulator #26: Way to disable light emulation?

Luigimeansme, 2021-11-20:
Is there a way to disable this emulator's function of emulating the lights on the toypad? It seems to cause problems with repeated keystone usage where if you use keystones too much and the colors keep changing, it bugs out and makes it so every time you switch the character you're moving around on the toypad, you have to drag them to the toybox then back to the toypad to get them to respond.

Luigimeansme, 2021-11-20:
[image]
Example of what the powershell says whenever the bug activates

cort1237, 2021-11-21:
I understand that the light emulation is a bit buggy. When I was writing I basically have to reverse engineer the effects, and at the time I didn't have an actual toypad with me, which made it harder. I will add an option to disable the light effects soon. 

As for the character switching errors that is actually unrelated to the light effects. Basically what was happening is the webpage stores all the character locations locally and seperately from the actual toypad emulator. And what was happening is somehow the webpage and toypad would get desynced. 

i.e. You moved a token to the top pad. Webpage knows this, but for whatever reason the toypad command didn't go through. So toypad still believes that token is in the old position. And now you have a desync.

But I have a better understanding of the library than I did when I wrote that code, and have implemented checks after certain actions that sync the webpage to exactly where the toypad thinks everyone is. And added a button to manually sync at the bottom of the page. So hopefully this should solve all those desync issues, cuz you weren't the only person experiencing them.

So follow the instructions in the read me to update and try it out.

Luigimeansme, 2021-11-22:
Just played through the entirety of Riddle-Earth 100% with no desync issues, that seems to have fixed it! Thanks Cort!

cort1237, 2021-11-22:
Ok great. I'm gonna close up this issue for now. I'll let you know when I make that light toggle.

closed 2021-11-22 by cort1237

Luigimeansme, 2021-11-22:
> Ok great. I'm gonna close up this issue for now. I'll let you know when I make that light toggle.

I don't need the light toggle anymore, I thought it was directly tied to the issue since it always happened whenever the lights changed. Knowing it's not, I'll leave the lights on, they're awesome