Berny23/LD-ToyPad-Emulator #34: Unable to scroll down while dragging

Luigimeansme, 2021-11-29:
I've noticed that while dragging a square at the top of the page, it will start to scroll up but it doesn't do the same when dragging a square at the bottom of the page. With the ToyPad being at the bottom of the page, I'd like to know if I'm doing anything wrong or if this is a known issue that is in the works

cort1237, 2021-11-29:
Nah you're not doing anything wrong. I know about that issue but thanks for pointing it out. Scroll boundaries for JQueryUI are really finicky and I struggled to get them fully working. Hope to fix that.

Berny23, 2021-11-29:
I already lowered the scroll speed some time ago, so you can scroll down on PC. But on phones and tablets there has to be a better solution.

Maybe you listen for the mouse move event and let the page scroll down if the user is dragging and the cursor is lower than a certain Y coordinate in the browser window. (like if height - cursorpos = 100).

Berny23, 2022-01-23:
I implemented a new scrolling mechanism in v1.3.4: https://github.com/Berny23/LD-ToyPad-Emulator/releases/tag/v1.3.4
Updating guide: https://github.com/Berny23/LD-ToyPad-Emulator#update

Luigimeansme, 2022-01-23:
Works great, no longer need to use a separate mouse with a dedicated scroll wheel!

closed 2022-01-23 by Luigimeansme