Berny23/LD-ToyPad-Emulator #42: Please connect the Toy Pad to the console.

ScenicGalaxy, 2022-01-23:
[image]
No matter what I try, I cant seem to get cemu to recognize the toy pad. Ive made sure to connect the emulator first, Im using Firefox on Windows. Ive even done the RPCS3 compatibility thing. Now what? any help would be appreciated

Berny23, 2022-01-23:
Please don't use the RPCS3 solution for Cemu, it breaks everything.
Now, you have to re-install the old driver:

1. Open device manager
2. Scroll down to USB devices and expand it
3. Double-click LEGO READER V2.10
4. Switch to the driver tab
5. Click previous driver
6. Select the first option and click yes
7. Restart Cemu

For Cemu, Zadig should look like this:
[image]

For RPCS3, Zadig should look like this:
[image]

Berny23, 2022-01-23:
I've added new troubleshooting sections, maybe they can help you too: https://github.com/Berny23/LD-ToyPad-Emulator#webpage-not-reachable-oracle-virtualbox

ScenicGalaxy, 2022-01-23:
Big help. It works. thanks

closed 2022-01-23 by ScenicGalaxy

yaser900, 2022-04-15:
I can't undo the changes on Zadig I tried doing that what you listed but it doesn't seem to be working the driver still is WinUSB and the game still doesn't connect to the toy pad can you help me please?

RetroZoneYT, 2022-08-16:
im also having this exact problem >_<

Nathy2705, 2022-11-12:
Ive got the exact same problem, hey berny can you help us?

lilkirill101, 2023-01-13:
I have a problem that I can't solve( Help me pleaseeeee
[image]
[image]

Berny23, 2023-01-13:
@lilkirill101 https://github.com/Berny23/LD-ToyPad-Emulator/issues/42#issuecomment-1019474653