Berny23/LD-ToyPad-Emulator #44: Suggestion: Add character and vehicle image toggle

Luigimeansme, 2022-01-25:
One feature that would make navigation easier is adding in a toggle between the current blue and green tiles with just the name and tiles with images on them instead. It's a lot easier to find what you're looking for based on visual design rather than reading the names.

Berny23, 2022-01-25:
Sorry, images won't come to this emulator because they are copyrighted material, and I don't want this repository to get a DMCA take-down. Everything on this repository must be my own work or that of the contributors.

closed 2022-01-25 by Berny23

damsleretour, 2023-04-10:
Hello, 

it's not possible just to add folder where put all images with names and each user will replace images as he want. You are in DMCA and all users have possibilities to replace images as he want.

If it's possible, i'm happy to see that. 

Thanks for all you do and for this beautiful software.

Respectually,

Damien.