Berny23/LD-ToyPad-Emulator #45: Suggestion: PS5 compatibility

VladimirKuletski, 2022-01-26:
Happy to share: works with PS5 (Pi Zero W).
Do you need any photos? Video?

Thanks once again for your work!

Berny23, 2022-01-26:
That's really nice! :)
I'll make a link to this page in the README file. You can post any video or images here, if you want to.

VladimirKuletski, 2022-01-26:
[image]

Attached a photo

VladimirKuletski, 2022-02-18:
@GlitcherFN 
Initially gone through the manual.
Started from PC. Connecting it to PC and check the ID of the emulated device.
Then connecting it to PS5. First was connecting it after the game was started. So it took several times to restart node or reboot to finally appear in the game.
Then I just added autostart to node app. And Pi is always connected to USB.  So while PS5 is starting it boots up and starts node. So, when I'm starting game it's already connected, booted. So game detects it immediately without any issues.

GlitcherFN, 2022-02-19:
Yeah i figured it out I had to start it before the game was open lol

GlitcherFN, 2022-02-24:
> @GlitcherFN
> Initially gone through the manual.
> Started from PC. Connecting it to PC and check the ID of the emulated device.
> Then connecting it to PS5. First was connecting it after the game was [...]

How did you add auto start to node?

VladimirKuletski, 2022-02-25:
Discussed here:
https://github.com/Berny23/LD-ToyPad-Emulator/discussions/51

closed 2022-03-24 by Berny23

Luigimeansme, 2024-01-25:
> I have everything setup it's just the website won't detect the game
> 
> @VladimirKuletski

Make sure you're using a Micro-USB DATA cable, not just one for charging.

VladimirKuletski, 2024-01-25:
Just to double check: try to connect (boot/start node application) it to PC and verify that it detects as ".. reader".
Might be same issues when USB emulation part is not configured correctly.

SONICq101010, 2024-01-25:
@Luigimeansme is the data one next to the HDMI

SONICq101010, 2024-01-25:
@VladimirKuletski how do I verify it?

SONICq101010, 2024-01-26:
@VladimirKuletski what do you mean?

SONICq101010, 2024-01-30:
@VladimirKuletski am I supposed to start the game before i start the pi or start the pi first?

SONICq101010, 2024-01-30:
@Luigimeansme how do I know if it's data or not

Luigimeansme, 2024-01-30:
> @Luigimeansme how do I know if it's data or not

If you have a phone that uses Micro-USB, plug it into your computer with that cable and see if it allows you to transfer files. If you're not prompted to transfer files, it's a charge only cable and you'll have to get a new one. Try to get one from Amazon, they'll say in the title if it's data supported or not, and they're very cheap

SONICq101010, 2024-01-31:
I think I have a data one

VladimirKuletski, 2024-01-31:
> @VladimirKuletski am I supposed to start the game before i start the pi or start the pi first?

Here is how I'm doing it:
1. Turning on PS5
2. Connecting device and waiting till it fully boots and starts program (I've added it to autostart)
3. Opening application website
4. Starting the game on PS5

Attached photo of "proper" port on Pi. I'm just connecting this to PS5
[image]

SONICq101010, 2024-01-31:
[image]

SONICq101010, 2024-01-31:
@VladimirKuletski it didn't work orangeballsmaster is my discord yes its weird but I just need help

VladimirKuletski, 2024-01-31:
> orangeballsmaster

[image]

Gravesurge, 2025-01-02:
I have been trying to figure this out, is there a video I can go through??

NovaAshwolfDev, 2025-01-03:
Damn this issue is still active? it's been two years

Gravesurge, 2025-01-03:
> Damn this issue is still active? it's been two years

just found this out, but i figured out my issue with the pi, now its saying unexpected syntax error

JonasBrinkmann-DE, 2025-05-09:
@Gravesurge What kind of syntax error do you get, can you share the full error? Maybe I can help.