Berny23/LD-ToyPad-Emulator #52: Syntax Error - How to fix?

SonicPrimus, 2022-02-14:
I was just playing LEGO Dimensions via CEMU and was using Benny, but when I took him off the ToyPad directly to the delete tab, all my toy tags disappeared and I got this error:

undefined:137
]25": 4026531839
 ^

SyntaxError: Unexpected number in JSON at position 3159
    at JSON.parse (<anonymous>)
    at initalizeToyTagsJSON (/home/sonicprimus/LD-ToyPad-Emulator/index.js:140:25)
    at Object.<anonymous> (/home/sonicprimus/LD-ToyPad-Emulator/index.js:27:1)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11

I have since tried to reconnect to the Emulator but get this same error. Is there a way to fix this?

Poley22, 2023-09-02:
help i was just trying to play LEGO DIMENSIONS on RPCS3 then this happend 
[image]

cort1237, 2023-09-03:
@Poley22 It's a syntax error in server/json/toytags.json.
There are three options:
1. Open the file and see if you can find the error. It follow a pretty easy to read format so you may be able to just spot it
2. Share the file here
3. Replace all contents with ```
{}
``` which will erase all tags in the toypad emulator. So you'll have to remake them/rewrite the vehicle tags.