Berny23/LD-ToyPad-Emulator #54: node index.js wont work

215phoenix, 2022-03-18:
I got this error
Error: Cannot find module 'node-ld'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
    at Function.Module._load (internal/modules/cjs/loader.js:591:27)
    at Module.require (internal/modules/cjs/loader.js:723:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/home/phoenix215/LD-ToyPad-Emulator/index.js:8:12)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
Can you help?

Berny23, 2022-03-24:
Did you run `cd LD-ToyPad-Emulator` before running `node index.js`?

Chapmike, 2022-03-25:
That's probably because there were some issues on step 5 while installing nmp for some days.
Try from this step again, it worked for me yesterday.

215phoenix, 2022-03-28:
To Berny23, I believe I did
To Chapmike can you give me the rundown?

215phoenix, 2022-03-28:
Also, I had a bigger error that may contribute to this error, I'll copy paste it if it comes up again

215phoenix, 2022-03-28:
I didn't get the error, and I did got it to work, Thanks Chap

closed 2022-03-28 by 215phoenix