Berny23/LD-ToyPad-Emulator #68: i cant install npm

The-BreadCat, 2022-05-25:
every time i try running npm install it just gets stuck on this message or something similar

[  ................] \ fetchMetadata: sill pacote range manifest for object-inspect@^1.9.0 fetched in 250ms

SlightlyPeckish, 2022-05-26:
[image]

hallinbirch, 2022-05-27:
dudes try my "Change one line" fix 
step Zero : open package.json in a IDE or Text editor\
step One :  change the line
"node-ld": "Berny23/node-ld",
To
"node-ld": "https://github.com/Berny23/node-ld.git",
Step Two :
try npm install again
It worked for me and i hope it works for you guys :)

The-BreadCat, 2022-05-30:
> dudes try my "Change one line" fix step Zero : open package.json in a IDE or Text editor step One : change the line "node-ld": "Berny23/node-ld", To "node-ld": "https://github.com/Berny23/node-ld.git", Step Two : try npm install again It worked for me and i hope it works for you guys :)

closed 2022-05-30 by The-BreadCat

ignacioat2, 2022-08-09:
> dudes try my "Change one line" fix step Zero : open package.json in a IDE or Text editor step One : change the line "node-ld": "Berny23/node-ld", To "node-ld": "https://github.com/Berny23/node-ld.git", Step Two : try npm install again It worked for me and i hope it works for you guys :)

bro how do i open package.json where i can find i in the computer or how can i access it?

hallinbirch, 2022-08-10:
this issue should have been fixed on june 6th just git the latest comments
and it should work
if you want to change  package.json it should be in the root of the source
i.e. ~/LD-ToyPad-Emulator/package.json or /home/(the username for the
vm)/LD-ToyPad-Emulator/package.json (wherever git cloned the
source)/package.json

commit 666bb8cb (2022-06-06): fix #68