Berny23/LD-ToyPad-Emulator #71: Suggestion: Rename tokens feature or edit Gollum's name

Luigimeansme, 2022-07-09:
This is INCREDIBLY minor, but I've been getting bothered by how Gollum's name on his tag goes past the boundaries of the tag. I'm suggesting either a feature to let you rename tags how you please, or changing his name so that it's on two layers rather than one, or just call him Gollum since that's what the game refers to him as. If this isn't a big enough issue to mess with, don't worry. I just wanted to post it and see what happens

Berny23, 2022-07-10:
I have already fixed this in my commit from April 22: https://github.com/Berny23/LD-ToyPad-Emulator/commit/1cd636ab4b498a4a5a5ff679d5a79d17b77dd57d

Please try to update to the latest version, then delete the Gollum tag from your toy box completely and add it again via the character selection.

Luigimeansme, 2022-07-10:
> I have already fixed this in my commit from April 22: [1cd636a](https://github.com/Berny23/LD-ToyPad-Emulator/commit/1cd636ab4b498a4a5a5ff679d5a79d17b77dd57d)
> 
> Please try to update to the latest version, then delete the Gollum tag from your toy box completely and add it again via the character selection.

My bad, I didn't notice this in a commit and I haven't deleted anyone from my toybox in a while! Thanks for letting me know!

closed 2022-07-10 by Luigimeansme