Berny23/LD-ToyPad-Emulator #74: Can't use Toy Pad emulator despite following all the steps provided.

wof8317, 2022-08-01:
I seem to have a problem with using this emulator with LD on either CEMU and RPCS3 on Linux. Maybe it's an issue with no proper udev rules provided to properly communicate with the device, much like how proper udev rules for dualshock 3/4 controllers are needed for rpcs3 to utilize said controllers. If you have a solution that is not what I mentioned, please reply as soon as you are able to. Thanks.

wof8317, 2022-08-01:
BTW: I'm using Arch Linux.

wof8317, 2022-08-02:
Nevermind, I seemed to solve the problem. It simply requires a simple udev rule to allow the use of the toypad emulator. This is actually required with the original Toy Pad in order to make it work on Linux. Here is my udev rule, it's gets stored as /etc/udev/rules.d/99-dimensions.rules (just rename the attached file to exclude the txt extension).
[99-dimensions.rules.txt](https://github.com/Berny23/LD-ToyPad-Emulator/files/9238065/99-dimensions.rules.txt). I'll see if this works with CEMU (it should work as well).

wof8317, 2022-08-02:
It doesn't seem to work with CEMU, probably due to a limitation with WINE and communicating with usb devices. Since CEMU doesn't have a native Linux port, I cannot test whether the udev rule works on it. Since the CEMU developers plan on open-sourcing the source code for the project sometime this year, I would rather just wait until someone successfully ports CEMU to Linux to perform further tests. For now, RPCS3 works with the Toy Pad emulator on Linux provided that the end-user provides a udev rule for the emulated usb device. So until the CEMU source code goes open, I will close the issue and reopen it at a later date.

wof8317, 2022-09-02:
Since Cemu went open source a week ago, I can't work on it right now, but I will reopen the issue just as a reminder to do so later this week.

Lazr1026, 2023-07-26:
Have you tested this out on your device yet? I have it all set up but cemu does not find the toypad.

wof8317, 2023-08-05:
I have and so far it doesn't work on cemu yet for Linux. However, support for USB devices, which would include this project, is something that the Cemu devs have planned for the linux port of Cemu. It's just a matter of time to implement it.

Lazr1026, 2023-08-05:
ah ok. thank yoy