Berny23/LD-ToyPad-Emulator #75: Pi Zero installation erroring

eggyclaw, 2022-08-06:
I'm performing a fresh install on a Pi Zero and getting some errors which isn't allowing me to run ToyPad.
I've used the new Raspberry Pi Imager to flash OS lite x64 and also using the advanced options to set up headerless install.

Stepping through the code line by line, everything seems fine until Step 5 and running 'npm install'.   When I run this I get the following in the SSH log:

> alex@raspberrypi:~/LD-ToyPad-Emulator $ npm install
> 
> > node-hid@2.1.1 install /home/alex/LD-ToyPad-Emulator/node_modules/node-hid
> > prebuild-install --runtime napi || node-gyp rebuild
> 
> prebuild-install WARN install No prebuilt binaries found (target=3 runtime=napi arch=arm64 libc= platform=linux)
> Package libusb-1.0 was not found in the pkg-config search path.
> Perhaps you should add the directory containing `libusb-1.0.pc'
> to the PKG_CONFIG_PATH environment variable
> No package 'libusb-1.0' found
> gyp: Call to 'pkg-config libusb-1.0 --cflags-only-I | sed s/-I//g' returned exit status 0 while in binding.gyp. whi                                                                                                                                                                                                          le trying to load binding.gyp
> gyp ERR! configure error
> gyp ERR! stack Error: `gyp` failed with exit code: 1
> gyp ERR! stack     at ChildProcess.onCpExit (/home/alex/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/lib/c                                                                                                                                                                                                          onfigure.js:259:16)
> gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
> gyp ERR! System Linux 5.15.32-v8+
> gyp ERR! command "/home/alex/.nvm/versions/node/v11.15.0/bin/node" "/home/alex/.nvm/versions/node/v11.15.0/lib/node                                                                                                                                                                                                          _modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /home/alex/LD-ToyPad-Emulator/node_modules/node-hid
> gyp ERR! node -v v11.15.0
> gyp ERR! node-gyp -v v8.4.1
> gyp ERR! not ok
> npm WARN ws@8.2.3 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yo                                                                                                                                                                                                          urself.
> npm WARN ws@8.2.3 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencie                                                                                                                                                                                                          s yourself.
> 
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! node-hid@2.1.1 install: `prebuild-install --runtime napi || node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the node-hid@2.1.1 install script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/alex/.npm/_logs/2022-08-06T11_13_55_847Z-debug.log

If I then try and run ToyBox I just get the following:

> alex@raspberrypi:~/LD-ToyPad-Emulator $ node index.js
> internal/modules/cjs/loader.js:670
>     throw err;
>     ^
> 
> Error: Cannot find module 'node-ld'
>     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
>     at Function.Module._load (internal/modules/cjs/loader.js:591:27)
>     at Module.require (internal/modules/cjs/loader.js:723:19)
>     at require (internal/modules/cjs/helpers.js:14:16)
>     at Object.<anonymous> (/home/alex/LD-ToyPad-Emulator/index.js:8:12)
>     at Module._compile (internal/modules/cjs/loader.js:816:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
>     at Module.load (internal/modules/cjs/loader.js:685:32)
>     at Function.Module._load (internal/modules/cjs/loader.js:620:12)
>     at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
> 

Any ideas on what could be happening?   The npm log suggests there are dependencies missing, could this be the case?

spudpiggy, 2022-08-07:
I'm not sure what causes this issue, but I know that if you force the install using `npm install --force` it works fine

eggyclaw, 2022-08-07:
Thanks - I can confirm using the force option did make the program work.   It's worth noting however, that the install process still outputted Warnings and Errors.

spudpiggy, 2022-08-24:
Change this issue to be about how you need to use force. That's still kind of a problem

UranVester, 2022-10-17:
sudo apt install libusb-1.0-0-dev

Run that before npm install, and no errors will be shown

commit 8ff98818 (2023-03-18): update installation guide; include package-lock.json file to fix dependency warnings and vulnerabilities; fixes #108, fixes #75
closed 2023-03-18 by Berny23