Berny23/LD-ToyPad-Emulator #82: Suggestion: Upload personal images for each token

Luigimeansme, 2022-10-18:
I understand we've discussed having images assigned to the characters and vehicles in #44, and completely understand not wanting to mess with copyright shenanigans. I did have an idea, however; add in the ability to upload images that you have on your device to assign them to the character or vehicle you're trying to create. Here's how I'm imagining this going down;
-You are making your characters for the first time by selecting them in the drop-down menu. When you click the "Create Character" or "Create Vehicle" button, a little button pops up asking if you'd like to upload an image. If you say yes, you get taken to your file finder and can select the image that'll be shown in the ToyBox. If you say no, it just goes into the ToyBox as it currently looks.

This way, we can choose whether or not we use images, and it's no risk to you because they have to be supplied by the user, and it doesn't even have to be the correct image. It can be any image you choose to represent that particular token

spudpiggy, 2022-12-16:
There should be some original images built in for those who don't own any LD figures and are too lazy to whip up a pictogram for each

Luigimeansme, 2022-12-16:
> @spudpiggy Links to **externally hosted** images would be OK, I think. But there will not be any copyrighted images uploaded to this repository. Big companies like Lego love to send DMCA take-downs.

I understood the copyright issue, that's why I suggested the option to upload images for your own personal version that you have to supply yourself. The original repository isn't edited or supplying copyrighted images, it's just the user wanting to customize their experience

VladimirKuletski, 2022-12-17:
There might be a logic that images will be expected in definite directory with predefined names/format/sizes.
If token image file exists webpage will render it, otherwise token will be presented "the old way" :)

So, anybody can handle their own collection without publishing it here.

spudpiggy, 2022-12-19:
> There might be a logic that images will be expected in definite directory with predefined names/format/sizes. If token image file exists webpage will render it, otherwise token will be presented "the old way" :)
> 
> So, anybody can handle their own collection without publishing it here.

You should also be able to add image links via the webui, which are then stored in a JSON file. You can then export/import the file using the site.

FrankTheTank908, 2023-06-22:
I would love to point to an img from a hoster to make it easier to figure out who they are talking about in the game. As I’m playing with my kids, I have to look at info and read the names and it’s time consuming. If I can see the image, that would make it so much easier to pick who they are talking about right away.

spudpiggy, 2023-06-22:
Why did you bump a 7 month old issue lmao

closed 2024-08-06 by Luigimeansme