Berny23/LD-ToyPad-Emulator #86: Alternatives for Raspberry Pi ?

SoldatRp11, 2022-10-24:
I was wondering if there is any alternative for the Raspberry ? Since there is a massive shortage of those currently. Sorry if this question was already asked, but I couldn’t manage to find it.

Berny23, 2022-10-24:
You can use any device with OTG USB support (this makes it act like a real Toy Pad) and a relatively recent Linux distribution (preferably Debian or Ubuntu).

TurboShell06, 2022-12-30:
> You can use any device with OTG USB support (this makes it act like a real Toy Pad) and a relatively recent Linux distribution (preferably Debian or Ubuntu).

And this won't work with virtual machines on a windows pc? I know it says VM is emulator only but i would like to play on real wii u since the emulators are so buggy with this game

raserdk, 2023-03-14:
i got the banana pi m2 zero to work with it 
with this image [Bananapi image](https://github.com/armbian/community/releases/download/202307/Armbian_23.02.0-trunk_Bananapi_lunar_edge_6.1.11.img.xz#bananapi%60)

dverzolla, 2023-05-02:
> i got the banana pi m2 zero to work with it with this image [Bananapi image](https://github.com/armbian/community/releases/download/202307/Armbian_23.02.0-trunk_Bananapi_lunar_edge_6.1.11.img.xz#bananapi%60)

Could you past here your full setup? Is it working well with LD-ToyPad-Emulator?

Regards.

jonahnm, 2023-10-27:
> You can use any device with OTG USB support (this makes it act like a real Toy Pad) and a relatively recent Linux distribution (preferably Debian or Ubuntu).

I wonder if this would work on android using LinuxDeploy?