Berny23/LD-ToyPad-Emulator #93: node index.js ERROR

headcrabzombie2, 2022-12-05:
[image]
getting this error when running node index.js for the last step tried closing terminal and going to the directory again but no luck any tips?

Berny23, 2022-12-16:
There have been no changes in the core program for 10 months. If you correctly follow the guide, everything works. I tested this numerous times on real hardware as well as in a virtual machine.

Luigimeansme, 2022-12-16:
I just completely redid my download a few months ago on a new SD and had no issues installing it. I recommend just copy/pasting the info into your command window rather than trying to type it out, key by key, as you open up the chance of errors with that.

spudpiggy, 2022-12-17:
I copied the commands on a fresh install. I always use SSH.

ovimihai, 2023-09-06:
I had the same error, the /dev/hidg0 was missing. I reinstalled from scratch, configured wifi from the Imager and just added the ssh file on the sd. I connected via ethernet wish ssh, ran all commands in order and it worked fine.

check after reboot the /dev/hidg0 exists