Berny23/LD-ToyPad-Emulator #97: Light blue is shown as green in the emulator

spudpiggy, 2022-12-16:
I use Dark Reader. Cba to get an image with it off. Here's a comparison image.
[image]

spudpiggy, 2022-12-16:
Oh yeah just realised that what is supposed to be f63bf8 is being shown as 76121c
and what should be edee32 is shown as 763e12

cort1237, 2023-01-14:
It's a known issue the colors aren't right. When they designed the toypad they worked around the bulb they knew they have and the diffusion of light caused by the plastic of the toypad. Because of this the RGB values sent by the game, are not accurate to the colors you'd see on the toypad. I tried a bunch of different manipulations but couldn't find one that accurately transformed the colors to what you see on the physical toypad.

I admittedly don't know much about colors, but anyone else wants to give it a shot. All color values go through "RGBToHex" function in "Index.js" so you could put any manipulations in there.

spudpiggy, 2023-01-16:
might have to go with a lookup table then, still its possible and should be done

Luigimeansme, 2023-01-17:
> might have to go with a lookup table then, still its possible and should be done

Maybe you should do it yourself then, rather than just assuming it's easy and expecting it to be done.

pablo5425, 2023-08-26:
> It's a known issue the colors aren't right. When they designed the toypad they worked around the bulb they knew they have and the diffusion of light caused by the plastic of the toypad. Because of [...]

i think the easiest workaround is simply grab the hex color and convert it to the actual desired value before its returned by the function.
i modified the function to show whats my plan:

`function RGBToHex(r, g, b) {
	r = r.toString(16);
	g = g.toString(16);
	b = b.toString(16);

	if (r.length == 1)
		r = "0" + r;
	if (g.length == 1)
		g = "0" + g;
	if (b.length == 1)
		b = "0" + b;
	var hex = "#" + r + g + b;
	
	switch(hex){
	case "#99240e":
		hex="#ffffff";
		break;
		
	default:
		break;
	}
	
	return hex;
}`

the idea would be add a case for each color we have to convert. this example code only converts #99240e to full white (what is supposed to be). there are not that many colors besides the 5th keystone so a switch is actually a good idea
any values not present in the switch would go through the default, displaying without any conversion

pablo5425, 2023-08-28:
@Berny23 i made a working color filter but push requests are disabled on my end. could you add the changed files?
[colorfilter.zip](https://github.com/Berny23/LD-ToyPad-Emulator/files/12457667/colorfilter.zip)