Berny23/LD-ToyPad-Emulator #104: I got 20 errors and 2 warnings and I don't know why

Brandonshev, 2023-01-24:
[image]

for context im using a Raspberry Pi Zero W, macbook pro and a real PS4.
Any help would be greatly appreciated 😊

joosthoi1, 2023-01-27:
Try running: "sudo apt install libusb-1.0-0-dev -y"

Brandonshev, 2023-01-27:
> Try running: "sudo apt install libusb-1.0-0-dev -y"

hi thank you for replying, i ran the command and it downloaded but sadly i still have the 2 errors, some things have changed, most noticeable is "`gyp` failed with exit code: 1" is now "`make` failed with exit code: 2" as seen below
[image]

closed 2023-02-14 by Brandonshev