Berny23/LD-ToyPad-Emulator #105: Xbox one support?

joosthoi1, 2023-01-27:
Hi, I tried this guide and after some troubleshooting (Was using a wrong cable without data smh and had to install USBLIB aswell) I got it working on my pc with Cemu. One quirk I found was that on pc I had to first power the pi zero W and only once it fully booted I could also add a cable to the data port and to my pc. I think this has something to do with it not switching to the right mode and windows not recognizing that. Maybe something that change in win 11. Also in device manager it didnt say that it was a lego reader v2.1 or what its supposed to be called, but as HID-compliant device. But it did work in CEMU.

Now to my actual question, I tried using it on xbox one but I couldnt get it working. Is there any thing I can change or are there tools I can use to debug the xbox one (x) USB ports/connections? Like is there a device manager for xbox one for example.
Also if there's any information about using the pi zero with xbox, please link it.

Or if this is working for someone else please tell me. Basically give me as much information if you can.

joosthoi1, 2023-01-27:
So I read that there are three different versions of the toypad, an xbox one version, an xbox 360 version and a multiplatform version for ps and wii u. Is there any information about the differences between these versions? I read that xbox controllers connected by usb use a security chip, does the toypad have this too? Can this too be emulated?

Luigimeansme, 2023-01-27:
Xbox platforms have a lot more hoops you have to jump through in order to create a compatible USB accessory, security chips and whatnot. It would be next to impossible for the emulator here to work on an Xbox One without a ton of modifications, likely some to the system itself to bypass the security restriction. Until then, the emulator is locked to emulators, Wii U, and Playstation systems only