Berny23/LD-ToyPad-Emulator #108: npm install fails on rpi zero W, with gyp error regarding libusb-1.0

ChristinMilloy, 2023-02-15:
[image]

ChristinMilloy, 2023-02-15:
Resolved by taking following steps:
1. Used rPi image "Raspberry Pi OS Lite (Legacy)" instead of "...OS Lite (32-Bit)" (this change may not be necessary).
2. Prior to `npm install` I did `sudo apt-get install libusb-1.0-0-dev` and `sudo apt-get install libudev-dev`

Perhaps the README should be updated to include this.

flashmasta, 2023-03-03:
That fixed the error when installing under Bullseye Lite (latest version of Raspberry Pi OS as of today)  You just need to run `sudo apt-get install libusb-1.0-0-dev` and `sudo apt-get install libudev-dev` before running `npm install`

DariusR10, 2023-03-10:
"npm WARN ws@8.11.0 requires a peer of bufferutil@^4.0.1 but none is installed. You must install peer dependencies yourself.
npm WARN ws@8.11.0 requires a peer of utf-8-validate@^5.0.2 but none is installed. You must install peer dependencies yourself."

After following the steps you guys gave and installing the npm it says i still need to install thoes 2 things.

How can thoes be installed?

and thank you for helping with this error, you are life saviors

commit 8ff98818 (2023-03-18): update installation guide; include package-lock.json file to fix dependency warnings and vulnerabilities; fixes #108, fixes #75
closed 2023-03-18 by Berny23