Berny23/LD-ToyPad-Emulator #117: I can't connect the pi to the real ps3

DariusR10, 2023-03-11:
I need someone to give a more detailed explanation on how to connect the pi with the ps3, when i try to, the emulated toy pad says that it can't see the game.

Luigimeansme, 2023-03-11:
Make sure you're using a micro USB data cable, like what you'd use to plug your phone into a computer to transfer files. Nothing else will work properly. If you don't have one, you can get one for a few bucks on Amazon

DariusR10, 2023-03-11:
> Make sure you're using a micro USB data cable, like what you'd use to plug your phone into a computer to transfer files. Nothing else will work properly. If you don't have one, you can get one for a few bucks on Amazon

That's what i use but it keeps saying the same. I even used a 2nd cable on the side usb port to give it more power but it still says the same.

I commented on the issue that was posted before this one about 2 programs that it WARNs me to install. Could it be because i haven't installed thoes programs that it doesn't work?

Bandud2, 2024-08-01:
Hey have you managed to do that?