Berny23/LD-ToyPad-Emulator #120: "Please Connect the ToyPad to the Console" Error on real Wii U hardware

EtienneGouraige, 2023-03-14:
I'm back. 

Attempting to see if I can solve this issue. Getting a "Please Connect the ToyPad to the Console" Error in the game on real Wii U hardware. I was successful in setting up the Pi Zero for this mod. No errors, and the Pi is running correctly. It just seems that this mod is not working on a real Wii U.

Any suggestions?

EtienneGouraige, 2023-03-14:
The Pi Zero W connects to the network just fine. The web page does load when I use my web browser to go into it. The "This message will disappear when I can see the game" red alert box on the web page doesn't disappear from the website when the Pi is plugged into my Wii U. I have also tried 3 USB cables (two I owned and one I _just_ got this morning) and attempted with all ports on my Wii U. Same error. I also tried unplugging my external hard drive.

I also tried using Taramisu instead of Aroma and reached the same error.

Any suggestions?

Luigimeansme, 2023-03-14:
> The Pi Zero W connects to the network just fine. The web page does load when I use my web browser to go into it. The "This message will disappear when I can see the game" red alert box on the web page doesn't disappear from the website when the Pi is plugged into my Wii U. I have also tried 3 USB cables (two I owned and one I _just_ got this morning) and attempted with all ports on my Wii U. Same error. I also tried unplugging my external hard drive.
> 
> I also tried using Taramisu instead of Aroma and reached the same error.
> 
> Any suggestions?
> 
> My next troubleshooting step is to simply attempt it Pi on a PS4/5.

Do you have any other devices that rely on micro usb data so you can test the cable? I had the same issue with mine when I first tried it, and my issue ended up being that none of my cables were actually data cables. Also make sure that the power cable is connected to the port closest to the edge, and the data cable that goes to the Wii U is plugged into the port further from the edge

EtienneGouraige, 2023-03-14:
Do you have any cable recommendations? I tested about 8 different cables and none of them worked. I have the power cable going into a wall outlet (right side, nearest to the corner) and the left is where I've been swapping the USB cables. 

Once I try a new cable, do I have to restart the game, or do I just wait for the notice to go away?

Luigimeansme, 2023-03-14:
> Do you have any cable recommendations? I tested about 8 different cables and none of them worked. I have the power cable going into a wall outlet (right side, nearest to the corner) and the left is [...]

https://www.amazon.com/gp/product/B0711PVX6Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
I use this exact cable for mine, so if this doesn't work it's something else. Once you try the new cable, the notice will go away as soon as it recognizes, restarting the game isn't necessary. The game will go from saying "Please connect the Toy Pad" to saying "No characters detected" to signify it sees the pad but nothing's on it

EtienneGouraige, 2023-03-14:
Thanks for the info.

That cable is the exact cable I ordered and it didn't work. I guess I'll go back from square one then.

Luigimeansme, 2023-03-14:
Try out the PS4 if you have it, that works for me but I never actually tested the Wii U cause I never needed to. I'll do some Wii U tests in the meantime, see if it works for me

Luigimeansme, 2023-03-14:
I just tried my Wii U copy with the emulator, just plugged into the front USB port, no issue at all. Can you send a picture of how you have your Pi plugged up, maybe there's an issue that we're not describing well?

EtienneGouraige, 2023-03-14:
[image]
[image]

Left wire is going to the front on the Wii U. Right is connected to Power and is going back to a wall outlet.

EtienneGouraige, 2023-03-15:
Got it working. 

I installed Pi Legacy version, then ran the following commands before running step 3.

sudo apt update
sudo apt install git
sudo apt-get install libusb-1.0-0-dev
sudo apt-get install libudev-dev

closed 2023-03-15 by EtienneGouraige