Berny23/LD-ToyPad-Emulator #126: How much RAM do I need?

DomanDragon, 2023-04-19:
I'm using a device that has 256MB of memory, and it's currently unable to work with installing nvm. If I switch the SD card to another device and install it on that, would I be able to run the emulator on a device with 256MB RAM?

Berny23, 2023-04-19:
Probably, but you must use the server version of Debian 11 (or any other Debian-based Linux distro with usb-gadget support, like Raspberry Pi OS Lite).

256MB is the minimum requirement, see: https://www.debian.org/releases/bullseye/amd64/ch03s04.en.html

DomanDragon, 2023-04-30:
The device I'm using in specific is a NanoPi NEO. I tried using the DietPi distro for it in combination with a USB Wi-Fi adapter, and it works.

closed 2023-04-30 by DomanDragon