Berny23/LD-ToyPad-Emulator #128: Question: is it possible to make such a toy pad emulator for the Skylanders Games?

Maceika, 2023-04-30:
Hello! so i've been searching through the internet for ways to emulate the skylanders figures and came upon this github that emulates the Lego Dimensions figures, i wanted to ask if there is any program or way to emulate the Skylander Figures? i do know that RPCS3 emulates them, but the FPS for the Skylanders games on RPCS3 is not good.. if anyone knows how to do it without RPCS3 then let me know, i would really appreciate it.

FrankTheTank908, 2023-06-24:
I will also be interested in this, my kids are loving the Lego one.  More options would be amazing. I’m enjoying this as well. Maybe more then my kids 😂

blan44, 2024-05-08:
Also wondering about the possibility of this, and if it could support Disney Infinity as well.
I know there’s Project D.I.R.E. but that is more complex and expensive to put together as it requires soldering and a custom PCB.