Berny23/LD-ToyPad-Emulator #131: LEGO READER V2.10 no longer showing

FuzzyLeo, 2023-05-03:
ive installed everything correctly, and even tried re-installing things just in case, but i can no longer get the "LEGO READER V2.10" to show up in virtualhere client, when i first installed everything it worked perfectly, however after leaving it for a while ive come back to it no longer functioning properly, it also can't connect to debian (not sure if you need virtualhere to be connected for it to work)

FuzzyLeo, 2023-05-03:
i went back and read a separate troubleshooting thing, and i found the issue

after doing the "Webpage not reachable (VMware)" troubleshooting option it worked, however the part that confused me (and potentially others) is in the instructions provided you would not find this out until after you have tried to launch the virtualhere client and find that its empty, leading people to try and fix the wrong issue.

FuzzyLeo, 2023-05-03:
update: so it turns out the reason it stopped working in the first place was because a bit ago i disabled the network adapters for a reason i cannot remember, after re-enabling that and setting everything else back what it was, it seems to finally be working.

closed 2023-05-04 by FuzzyLeo

MilesTailsFan2005, 2023-11-13:
You Could Also Reinstall The Device If Your Using Virtual Box
Make Sure To Have Bridge Adapter Enabled On Network