Berny23/LD-ToyPad-Emulator #134: SyntaxError: Unexpected token } in JSON at position 18873

farudel, 2023-05-06:
Hi, how can fix this error?
undefined:794
]   }
    ^

SyntaxError: Unexpected token } in JSON at position 18873
    at JSON.parse (<anonymous>)
    at initalizeToyTagsJSON (/home/farudel07/LD-ToyPad-Emulator/index.js:140:25)
    at Object.<anonymous> (/home/farudel07/LD-ToyPad-Emulator/index.js:27:1)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11

cort1237, 2023-05-06:
There's a formatting error in the toytags.json file in server/JSON/

The entire JSON file is contained with square braces [], and each toytag data entry is contained inside by curly braces {}. For some reason the square braces closed before all the curly braces.

If you can share the file here, but you may be able to fix it yourself by just finding the misplaced square closing brace.

cort1237, 2023-05-07:
Or there's just an unneeded `}` at the end of the file you can delete.