Berny23/LD-ToyPad-Emulator #137: why

luckip1, 2023-05-23:
I did everything, but the site says:
WARNING: Do not place any characters, I have not communicated with the game yet.
Placing characters now will cause me to crash.
This message will disappear when I can see the game.
What should I do?

dverzolla, 2023-05-23:
Send more information.
Did you open the game? Which emulator? etc.

luckip1, 2023-05-24:
yes, i opened the game. I am using cemu emulator

luckip1, 2023-05-24:
[image]

luckip1, 2023-06-01:
help me pls

dverzolla, 2023-06-09:
For some reason you are missing something. Delete everything, re read the README.md and try from scratch again.

SiemJanssen, 2023-07-03:
I also seem to be having this issue, although it has worked at first it doesn't anymore. I am using rpcs3. It doesn't seem to connect with the game somehow

luckip1, 2023-08-01:
> I also seem to be having this issue, although it has worked at first it doesn't anymore. I am using rpcs3. It doesn't seem to connect with the game somehow
> 
> EDIT: I fixed it by using Zadig to change the driver of the portal to WinUSB

how to use it?