Berny23/LD-ToyPad-Emulator #153: MacOS Cemu can't connect to the Toy Pad

HalLeopern, 2023-09-09:
I'm on MacOS, and everything seems to be working, except Cemu can't detect the game, and the Emulator can't detect Cemu. What should I do?
[image]

dverzolla, 2023-09-12:
Does virtualhere is working properly? If so, check this out: https://github.com/Berny23/LD-ToyPad-Emulator#virtualhere-shows-lego-reader-v210-but-fails-with-operation-not-permitted

HalLeopern, 2023-09-12:
Yeah, virtualhere is working perfectly

dverzolla, 2023-09-12:
In windows it's pretty straightforward. If VirtualHere shows "LEGO READER", so the game detects the ToyPad.

pablo5425, 2023-09-14:
make sure "LEGO READER 2.0" is connected. you have to double click it to plug it in. also emulator cant be open while you do that
if it still doesnt work then i guess it needs some special drivers for mac

deReeperJosh, 2024-07-07:
MacOS Cemu didn't have usb support until recently so you should test this again