Berny23/LD-ToyPad-Emulator #156: Npm install fails every time on Raspberry Pi Zero W

LoneNomadBlue, 2023-10-10:
Hello I have failed to get this to work with 40+ attempts :(

I used Option 1 with a Virtual Machine and it worked perfectly. It was easy to setup thanks to the setup video and I successfully used it with CEMU/RPCS3. I wanted to try the virtual machine option first so that I knew it worked before buying a raspberry pi.

Since it worked I went ahead and bought a Raspberry Pi Zero W. I have multiple USB Type-A to micro-USB 2.0 Type-B cable that supports data transmission. I tested each of them to make sure that they actually can transfer data. I have a 64gb SD card.

I installed Raspberry Pi OS Lite (32-Bit) / Raspberry Pi OS Lite (64-Bit) / Raspberry Pi OS Lite (Legacy) using the Raspberry Pi Imager tool. No matter which one I tried all of them successfully installed to the SD card and worked with the Raspberry Pi Zero W. But no matter which one I installed I never got the program to work with either version. The logs I posted below are from Raspberry Pi OS Lite (Legacy) version. IDK if it even makes a difference. I make sure to configure wireless lan, enable ssh, set username-password, set hostname, set wireless lan country etc.

I connect the pi to my PC using the Micro-USB Data port, the one labeled USB not PWR. Just like the one in the [ First demo video on Cemu emulator: https://www.youtube.com/watch?v=7CBa9u2ip-Y ] I wait until my pi properly boots since the first boot takes a bit longer. I successfully connect via SSH using windows terminal by typing [ ssh abisselfar@192.168.50.54]. I enter my password and it connects.

I click the copy code button on step 3 -> [Use SSH to run the following commands] and then right click in terminal to paste it. I think it doesn't install git? It just fails when the [ Do you want to continue? [Y/n] ] part comes up when you enter [ sudo apt install -y git libusb-1.0-0-dev libudev-dev ]. According to this person https://github.com/Berny23/LD-ToyPad-Emulator/issues/109#issuecomment-1445149259 he solved this issue by replacing "sudo apt install -y git" in Step 3's command code with the following code:

sudo apt update
sudo apt install git

so that's what I did. Now it pauses and asks for [Y/n] which I then enter " Y ".

If I don't do this later on it fails at these commands

[
git config pull.rebase false
git clone https://github.com/Berny23/LD-ToyPad-Emulator.git
cd LD-ToyPad-Emulator
]

After this bit it's just as shown below. I reboot the pi. I reconnect using SSH. I click the copy code button for Step 5. -> [Connect via SSH again and run the following commands]. I right click in the terminal to paste the commands and it fails at [ npm install ] as seen below. This seems to be one of the two types of failures that usually happen. Either I get this error as shown below or I get another error where it installs but there are 3 vulnerabilities and it says that I can do [ npm audit fix ] or [npm audit]. If I do [ npm audit fix ] it fixes 2 of the vulnerabilities but then it tells me that the last vulnerability needs a manual review. Idk what to do.

I've tried copy pasting each line individually. I tried copy pasting all the commands of each step together. I tried copying in groups until where there are blank lines. I tried connecting to SSH from Powershell. I tried connecting to SSH from the Debian virtual machine. Doesn't change anything I still get one of the two types of errors logs. I tried the 3 different lite versions of Pi OS as I've mentioned above. I tried using multiple different cables. I've tried the solutions from https://github.com/Berny23/LD-ToyPad-Emulator/issues/25 and https://github.com/Berny23/LD-ToyPad-Emulator/issues/21 and https://github.com/Berny23/LD-ToyPad-Emulator/issues/109 and https://github.com/Berny23/LD-ToyPad-Emulator/issues/120 but it seems that these issues were already fixed and the commands were updated.

I guess I'm way over my head because I have no idea at this point. Even though the server runs as usual it never recognizes that the game is running and the emulator is stuck on the "Connect the Toypad" screen since it never recognizes the Pi. It has something to do with the [ npm install ] failing but idk why it keeps failing.

I've been trying back to back for 2 days now. I'll try and see if I can get it to fail the other way with 3 vulnerabilities tomorrow and I'll post the log again.

I appreciate any help. I'm totally clueless as to what's going wrong and I don't understand the errors it's giving. Sorry for the long wall of text. If someone could please help. Thank you.

```Microsoft Windows [Version 10.0.22621.2361]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Abisselfar>ssh abisselfar@192.168.50.54
The authenticity of host '192.168.50.54 (192.168.50.54)' can't be established.
ED25519 key fingerprint is SHA256:t6pZM4huZOeRknOX9Q5u9tRLxyyD5kj9IYXRSmDufP0.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.50.54' (ED25519) to the list of known hosts.
abisselfar@192.168.50.54's password:
Linux toypad 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
abisselfar@toypad:~ $ sudo apt install git libusb-1.0-0-dev libudev-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  git-man libcurl3-gnutls liberror-perl libusb-1.0-doc
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
Need to get 6,135 kB of archives.
After this operation, 33.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://mirror.as43289.net/raspbian/raspbian buster/main armhf libcurl3-gnutls armhf 7.64.0-4+deb10u6 [293 kB]
Get:2 http://mirror.as43289.net/raspbian/raspbian buster/main armhf liberror-perl all 0.17027-2 [30.9 kB]
Get:3 http://mirror.as43289.net/raspbian/raspbian buster/main armhf git-man all 1:2.20.1-2+deb10u8 [1,623 kB]
Get:4 http://mirror.as43289.net/raspbian/raspbian buster/main armhf git armhf 1:2.20.1-2+deb10u8 [4,187 kB]
Fetched 6,135 kB in 4s (1,703 kB/s)
[... 246 lines cut ...]
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
gyp: Call to 'pkg-config libusb-1.0 --cflags-only-I | sed s/-I//g' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/home/abisselfar/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/lib/configure.js:259:16)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 5.10.103+
gyp ERR! command "/home/abisselfar/.nvm/versions/node/v11.15.0/bin/node" "/home/abisselfar/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/abisselfar/LD-ToyPad-Emulator/node_modules/node-hid
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v8.4.1
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@2.1.2 install: `prebuild-install --runtime napi || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@2.1.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/abisselfar/.npm/_logs/2023-10-10T21_06_49_126Z-debug.log
abisselfar@toypad:~/LD-ToyPad-Emulator $```

LoneNomadBlue, 2023-10-10:
Here is the other error that I usually get that I mentioned before. This happens if I don't change the command from step 3 [ sudo apt install -y git libusb-1.0-0-dev libudev-dev ] to [ sudo apt install git libusb-1.0-0-dev ] as explained in https://github.com/Berny23/LD-ToyPad-Emulator/issues/109#issuecomment-1445149259

If you look at the log below you can see that it just skips the git commands in step 3 

[
git config pull.rebase false
git clone https://github.com/Berny23/LD-ToyPad-Emulator.git
cd LD-ToyPad-Emulator
]

It never git clones the repository so it can't change directory into it. Because of this I end up creating the repo myself and rerunning [ npm install ] as you can see below. This obviously messes some other stuff up but I have no idea what's actually going wrong here.

I'll appreciate any help. I really want to get this to work because I really love Lego Dimensions and want an actual physical Toy Pad Emulator for it instead of just a virtual machine. Please help me. Thank you. 

```Microsoft Windows [Version 10.0.22621.2361]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Abisselfar>ssh abisselfar@192.168.50.54
The authenticity of host '192.168.50.54 (192.168.50.54)' can't be established.
ED25519 key fingerprint is SHA256:D+HrN9BDTdYsDPROZklqR7dwIWXHMwKruNBiB5SyfLY.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.50.54' (ED25519) to the list of known hosts.
abisselfar@192.168.50.54's password:
Linux toypad 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
abisselfar@toypad:~ $ sudo apt update
toverlay=dwc2" | sudo tee -a /boot/config.txt
echo "dwc2" | sudo tee -a /etc/modules
echo "libcomposite" | sudo tee -a /etc/modules
echo "usb_f_rndis" | sudo tee -a /etc/modules

git config pull.rebase false
git clone https://github.com/Berny23/LD-ToyPad-Emulator.git
cd LD-ToyPad-Emulator

printf '\necho "$UDC" > UDC\nsleep 2;\nchmod a+rw /dev/hidg0' >> usb_setup_script.sh

sudo cp usb_setup_script.sh /usr/local/bin/toypad_usb_setup.sh
sudo chmod +x /usr/local/bin/toypad_usb_setup.sh
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]_usb_setup.sh") | sudo crontab -
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [400 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [110 kB]
Selecting previously unselected package liberror-perl.
Preparing to unpack .../1-liberror-perl_0.17027-2_all.deb ...
Unpacking liberror-perl (0.17027-2) ...
Selecting previously unselected package git-man.
Preparing to unpack .../2-git-man_1%3a2.20.1-2+deb10u8_all.deb ...
Unpacking git-man (1:2.20.1-2+deb10u8) ...
Selecting previously unselected package git.
Preparing to unpack .../3-git_1%3a2.20.1-2+deb10u8_armhf.deb ...
Unpacking git (1:2.20.1-2+deb10u8) ...
Selecting previously unselected package libudev-dev:armhf.
Preparing to unpack .../4-libudev-dev_241-7~deb10u9+rpi1_armhf.deb ...
Unpacking libudev-dev:armhf (241-7~deb10u9+rpi1) ...
Selecting previously unselected package libusb-1.0-0-dev:armhf.
Preparing to unpack .../5-libusb-1.0-0-dev_2%3a1.0.22-2_armhf.deb ...
Unpacking libusb-1.0-0-dev:armhf (2:1.0.22-2) ...
Selecting previously unselected package libusb-1.0-doc.
Preparing to unpack .../6-libusb-1.0-doc_2%3a1.0.22-2_all.deb ...
Unpacking libusb-1.0-doc (2:1.0.22-2) ...
Setting up libusb-1.0-doc (2:1.0.22-2) ...
[... 303 lines cut ...]
│                                                                              │
│         Visit https://go.npm.me/audit-guide for additional guidance          │
└──────────────────────────────────────────────────────────────────────────────┘
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Moderate      │ semver vulnerable to Regular Expression Denial of Service    │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ semver                                                       │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Patched in    │ >=5.7.2                                                      │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ 1b4b7b380f45e3c6c69576e2177741be758ff47840c398868beb49d4696… │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ 1b4b7b380f45e3c6c69576e2177741be758ff47840c398868beb49d4696… │
│               │ > usb > prebuild-install > node-abi > semver                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://github.com/advisories/GHSA-c2qf-rxjj-qqgw            │
└───────────────┴──────────────────────────────────────────────────────────────┘
found 3 moderate severity vulnerabilities in 184 scanned packages
  run `npm audit fix` to fix 2 of them.
  1 vulnerability requires manual review. See the full report for details.
abisselfar@toypad:~/LD-ToyPad-Emulator $ npm audit fix
updated 1 package in 36.924s
fixed 2 of 3 vulnerabilities in 184 scanned packages
  1 vulnerability required manual review and could not be updated
abisselfar@toypad:~/LD-ToyPad-Emulator $```

cort1237, 2023-10-11:
The first one is failing because libusb doesn't ever install properly. The second one fails because you deleted the libusb line, again not installing right and cut out an echo with it breaking the command.

From your second comment:
> abisselfar@toypad:~ $ sudo apt update
toverlay=dwc2" | sudo tee -a /boot/config.txt
echo "dwc2" | sudo tee -a /etc/modules
echo "libcomposite" | sudo tee -a /etc/modules
echo "usb_f_rndis" | sudo tee -a /etc/modules
> 
> git config pull.rebase false
git clone https://github.com/Berny23/LD-ToyPad-Emulator.git
cd LD-ToyPad-Emulator

toverlay=dwc2" is both not a command and a broken string, so it's interrupting.

But ultimately these logs feel like a mashup of different attempts, we need a baseline. Reimage again to just the normal 32-bit Raspberry Pi Lite in the Raspberry Pi Imager. Run each command line-by-line one at a time, copy and pasted. And then share those logs and we can figure out what happened.

LoneNomadBlue, 2023-10-11:
It's not that I delete the libusb line it just never appears on the console. When I copy paste the entire line of commands for step 3 it just doesn't show up. Like below I have the log for copy pasting every single line one by one but it still doesn't show up. I did this whole process 3 times because I just could not figure out what was happening to the second command which is [ sudo apt install -y git libusb-1.0-0-dev libudev-dev ]. Like I copy and paste the line but then after it starts doing it the line when I entered it as [ abisselfar@toypad:~ $ sudo apt install -y git libusb-1.0-0-dev libudev-dev ] just disappears. The results from that command are there but me entering it is not. I thought I was gaslighting myself at first so I just stared at the screen the entire time during my second try. Then I still tried a third time and this log is from the third try. This part below is the response from the terminal to the second command which is -> [ sudo apt install -y git libusb-1.0-0-dev libudev-dev ].

```
Preparing to unpack .../0-liberror-perl_0.17029-1_all.deb ...
Unpacking liberror-perl (0.17029-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../1-git-man_1%3a2.30.2-1+deb11u2_all.deb ...
Unpacking git-man (1:2.30.2-1+deb11u2) ...
Selecting previously unselected package git.
Preparing to unpack .../2-git_1%3a2.30.2-1+deb11u2_armhf.deb ...
Unpacking git (1:2.30.2-1+deb11u2) ...
Selecting previously unselected package libudev-dev:armhf.
Preparing to unpack .../3-libudev-dev_247.3-7+rpi1+deb11u2_armhf.deb ...
Unpacking libudev-dev:armhf (247.3-7+rpi1+deb11u2) ...
Selecting previously unselected package libusb-1.0-0-dev:armhf.
Preparing to unpack .../4-libusb-1.0-0-dev_2%3a1.0.24-3_armhf.deb ...
Unpacking libusb-1.0-0-dev:armhf (2:1.0.24-3) ...
Selecting previously unselected package libusb-1.0-doc.
Preparing to unpack .../5-libusb-1.0-doc_2%3a1.0.24-3_all.deb ...
Unpacking libusb-1.0-doc (2:1.0.24-3) ...
Setting up libusb-1.0-doc (2:1.0.24-3) ...
Setting up libusb-1.0-0-dev:armhf (2:1.0.24-3) ...
Setting up liberror-perl (0.17029-1) ...
Setting up udev (247.3-7+rpi1+deb11u2) ...
Setting up libudev-dev:armhf (247.3-7+rpi1+deb11u2) ...
Setting up git-man (1:2.30.2-1+deb11u2) ...
Setting up git (1:2.30.2-1+deb11u2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for initramfs-tools (0.140) ...
```

I just have no idea why that line of me entering the command disappears. If you want I can record the entire process of me installing the Pi OS using the imager tool and then connecting via SSH and entering each command line by line.

Right below is the entire terminal line.

```
Microsoft Windows [Version 10.0.22621.2361]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Abisselfar>ssh abisselfar@192.168.50.54
The authenticity of host '192.168.50.54 (192.168.50.54)' can't be established.
ED25519 key fingerprint is SHA256:JE84l2qHzf+w/EvnmU6h9LjuGrQagKQHsgc88xh09B4.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.50.54' (ED25519) to the list of known hosts.
abisselfar@192.168.50.54's password:
Linux toypad 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
abisselfar@toypad:~ $ sudo apt update
Get:1 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]
Get:4 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [314 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian bullseye/non-free armhf Packages [106 kB]
Fetched 13.7 MB in 24s (560 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Preparing to unpack .../0-liberror-perl_0.17029-1_all.deb ...
Unpacking liberror-perl (0.17029-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../1-git-man_1%3a2.30.2-1+deb11u2_all.deb ...
Unpacking git-man (1:2.30.2-1+deb11u2) ...
[... 250 lines cut ...]
  COPY Release/HID_hidraw.node
make: Leaving directory '/home/abisselfar/LD-ToyPad-Emulator/node_modules/node-hid/build'

> bufferutil@4.0.7 install /home/abisselfar/LD-ToyPad-Emulator/node_modules/bufferutil
> node-gyp-build

make: Entering directory '/home/abisselfar/LD-ToyPad-Emulator/node_modules/bufferutil/build'
  CC(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
make: Leaving directory '/home/abisselfar/LD-ToyPad-Emulator/node_modules/bufferutil/build'

> utf-8-validate@5.0.10 install /home/abisselfar/LD-ToyPad-Emulator/node_modules/utf-8-validate
> node-gyp-build

make: Entering directory '/home/abisselfar/LD-ToyPad-Emulator/node_modules/utf-8-validate/build'
  CC(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/home/abisselfar/LD-ToyPad-Emulator/node_modules/utf-8-validate/build'
added 184 packages from 246 contributors and audited 184 packages in 417.321s
found 3 moderate severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
abisselfar@toypad:~/LD-ToyPad-Emulator $
```

LoneNomadBlue, 2023-10-11:
This time the I recorded the process and uploaded it to Youtube as an unlisted video. I sped the video up since it became too long, it's still pretty slow and you can speed down the video if necessary or I can reupload at original speed. I actually recorded a couple of takes but this final one worked pretty well because the [ abisselfar@toypad:~ $ sudo apt install -y git libusb-1.0-0-dev libudev-dev ] command actually stayed on console instead of disappearing after I execute it. I still have no idea why it does that when I copy paste it directly and the output stays the same no matter what happens anyways. There is also a picture of how I plug the Pi in. I just tested it again to make sure the cable can transmit data and it successfully transferred data from my old phone to the PC. The port is correct it's the USB one not the PWR one. I individually copy paste each line. I actually copied the [ git config pull.rebase false ] command again after git cloning and changing the directory to the repo but no output on the terminal again. Idk if that's actually what is supposed to happen. Once again 3 vulnerabilities after the [ npm install ] command. Idk what to do but thanks for trying to help. I hope I can get this to work. It's like the 60+ attempt at this point.

Process Video: https://www.youtube.com/watch?v=kZDiiFdW2F8 

[image]

LoneNomadBlue, 2023-10-11:
Yeah I tried it so many times. As I've said before the game is stuck on the connect toypad screen and the webpage never updates from 

WARNING: Do not place any characters, I have not communicated with the game yet.
Placing characters now will cause me to crash.
This message will disappear when I can see the game.

I have tried the game with the virtual machine as I've said before and it works perfectly fine using VMware. It updates the page to remove the warning when I open the emulator and get in game. Not on the pi though :(

LoneNomadBlue, 2023-10-11:
Yes I tried it without audit fix and I tried it with audit fix. I run the toypad server first before launching Cemu and then launching the game. I just followed the First demo video on Cemu emulator: https://www.youtube.com/watch?v=7CBa9u2ip-Y on the repository page. The server does run and I can access the webpage. It syncs tags as well. The only problem is the game doesn't see the Pi as the toypad. I have connected it to my PC using the correct port and a cable that can transfer data which I did check by actually transferring data. The only thing that I see different from what other people have done is that my installation either ends up getting 3 vulnerabilities or the Usb driver doesn't install properly as you've said. So I don't know what else can e the issue other than this.

After installing it for 80+ times I can safely say that I just keep running into the same two problems that I've written about before. Either I run it line by line and the npm install shows 3 vulnerabilities and it doesn't recognize the Pi as a usb gadget. Or if I execute the commands by copy pasting the whole step instead of line by line then it fails to install the usb driver and npm install also gives various ERR lines and once again it doesn't recognize the Pi as a usb gadget.

cort1237, 2023-10-11:
Again I'm telling you you're not "gaining vulnerabilities" everyone gets those it's just DDoS exploits found in the specific versions for a few libraries we use, but this is not an internet device so that does not matter. They are not an error or issue. Your scenario one is describing a successful installation. I even tested a fresh install on my Pi and had success.

But it leaves me kinda stumped if everything you say is true the only thing that makes sense to me is something is wrong with the USB connection. Either your usb cable, your computer, or the Pi itself.

LoneNomadBlue, 2023-10-11:
@cort1237 IT WORKS!!! :D I'M SO HAPPY AND SO GLAD THAT IT WORKS!!! I have absolutely no idea why it started working. I literally changed nothing with the Pi or the cable or the port or the OS version or anything else. The only thing that I changed was to launch the US version of the game on CEMU. See up until now I had only launched the EU version of Lego Dimensions on CEMU and RPCS3 and every single time it failed to recognize the Pi as the ToyPad. This was odd because on the virtual machine it worked with the EU versions of the game on both emulators. Even on @Berny23 's Youtube videos in the comment section he had posted this:

FAQ:
⇾ Passwords on Linux (Debian) are invisible while typing, that's normal. Just type it (in my video it's “123”) then press the Enter key.
⇾ The Wii U version I'm using has the ID "APZPWR".
⇾ The PS3 version I'm using has the ID “BLES02105”.
⇾ In the video, I'm using the EU version that contains all languages of the game, including English. I didn't test US, but it should work too.
⇾ For PS3: All 16 DLC files & the latest game update 1.24 must be installed for all worlds to be accessible. I WILL NOT PROVIDE THESE FILES.
⇾ For Wii U: The complete DLC package & the latest game update 352 must be installed for all worlds to be accessible. I WILL NOT PROVIDE THESE FILES.

And just so you know I have both of these with the latest update and the all of the DLC. For CEMU the EU version is APZPWR with update version 352 and DLC version 112. For RPCS3 the EU version is BLES02105 with update version 1.24 and 16 DLC's. I know there's no issue with the Emulators or the Game/Update/DLC files because they already work with the Toypad emulator using the virtual machine on VMware. I even redownloaded and reinstalled the Emulators and Game/Update/DLC files just in case but nothing fixed it.

Then I just randomly decided to try the US version which is APZEWR with update version 320 and DLC version 128 and to my surprise it just worked. The warning went away on the server webpage and the "Connect the Toypad" screen went away in game. I tested to see if the characters worked and they did. After that I immediately exited the US version and opened the EU version and now the EU version also worked.

I have no idea how or why the US version worked and I especially have no idea how launching the US version fixed the EU version and made it work now on both emulators. But I am quite satisfied that it works. Sorry for taking your time and thank you for all the help! Now I'm gonna build a custom Toypad out of Lego to put my Raspbery Pi inside.

cort1237, 2023-10-11:
Good to hear. Hope you have fun

closed 2023-10-11 by cort1237