Berny23/LD-ToyPad-Emulator #157: Lego Reader doesn't appear in vhui64

adriiba, 2023-10-17:
I have done all the steps and in principle everything is fine until I open the virtual here client, and the desktop hub with the lego reader does not appear, I have checked the host and looked for different solutions but none of them make it appear. Any possible solution? I even provided a help window on github that said I had to write some kind of command when the lego reader didn't appear, but neither did it. help pls

ElBread3, 2023-10-21:
Oh there's a part in the Troubleshooting section near the bottom of the readme, this might be of help

### VirtualHere USB Client doesn't show LEGO READER V2.10
When installing the virtual machine, you have to set the hostname to `debian`.

Alternatively, copy the following command and replace `YOUR_IP_ADDRESS` with your virtual machine's IP address (it looks like `192.168.X.X`, run `hostname -I` to show it). After you've done this, run the modified command while you're inside the `LD-ToyPad-Emulator` folder.
````bash
git reset --hard ; printf '\necho "usbip-vudc.0" > UDC\nusbipd -D --device\nsleep 2;\nusbip attach -r YOUR_IP_ADDRESS -b usbip-vudc.0\nchmod a+rw /dev/hidg0' >> usb_setup_script.sh ; sudo cp usb_setup_script.sh /usr/local/bin/toypad_usb_setup.sh
````
in case you haven't seen this already.

randomoOCoder, 2024-08-25:
Hi, I have the same problem. I made sure the hostname of the VM was '**debian**' and ran the command just to be sure, but it didn't work. I also set the adapter to "**bridged adapter**" in the virtual box settings (I selected the same WiFi device as my host machine, I assume that's correct ?).  I'm running **linux mint 22** on the host.

Wonkrahv, 2025-02-19:
Instructions unclear do i copy the ur ip adress command into the terminal or so i put my actual ip  and what modified command and where bru