Berny23/LD-ToyPad-Emulator #164: not working on cemu anymore

pablo5425, 2023-12-16:
after some windows updates this thing stopped connecting with wii u
server works, virtualhere works and i can plug in toypad without errors but emulator just wont recognize it, acting like nothing was connected

pablo5425, 2023-12-16:
a vmware update apparently changed some settings. i changed network adapter back to bridged but there is still no connection. im not sure if the problem is on cemu or if toypad emulator just cant see the game

pablo5425, 2023-12-16:
even after a fresh install it doesnt seem to work anymore. it just cant see the game

pablo5425, 2023-12-16:
ok i found out what was wrong
**CAREFUL WITH UPDATING WINDOWS 11 TO 23H3**. apparently it replaced my drivers with winusb (for rpcs3)

steps to revert it (i use windows in spanish. expect inaccurate translations):
1. go to device manager (zadig wont install HID drivers)
2. go to USB devices > LEGO READER V2.10
3. right click > update driver
4. examine pc
5. pick from a list of available drivers
6. choose "USB INPUT DEVICE" (the only one that does not say LEGO READER)

this will change it to HID drivers. you wont be able to easily find the toypad in device manager anymore bc it changes to a generic name but zadig will display it as LEGO READER V2.10 whenever you want to change drivers for rpcs3

danylopez123, 2024-02-10:
> ok i found out what was wrong
> ### **CAREFUL WITH UPDATING WINDOWS 11 TO 23H2**.
> 
> apparently it replaced my drivers with winusb (for rpcs3)
> 
> steps to revert it (i use windows in spanish. expect inaccurate translations):
> 
>     1. go to device manager (zadig wont install HID drivers)
> 
>     2. go to USB devices > LEGO READER V2.10
> 
>     3. right click > update driver
> 
>     4. examine pc
> 
>     5. pick from a list of available drivers
> 
>     6. choose "USB INPUT DEVICE" (the only one that does not say LEGO READER)
> 
> 
> this will change it to HID drivers. you wont be able to easily find the toypad in device manager anymore bc it changes to a generic name but zadig will display it as LEGO READER V2.10 whenever you want to change drivers for rpcs3

I was also in the same spot after my test with RPCS3 then wanted to go back to Cemu, thanks for the fix!