Berny23/LD-ToyPad-Emulator #173: Help!!

Brezenpilot, 2024-01-25:
Download:1 http://security.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Found:2 http://deb.debian.org/debian bookworm InRelease
Found:3 http://deb.debian.org/debian bookworm-updates InRelease
Recovered 48.0 kB in 1s (67.9 kB/s)
Reading package list... Done
Generating dependency tree... Done
Reading status information... Done
All packages are updated.
Reading package list... Done
Generating dependency tree... Done
Reading status information... Done
The python package has no versions available, but is named by another
package. This may indicate that the package is missing, obsolete
or it is only available within another source
However these packages replace it:
    python-is-python3 2to3

E: The "python" package has no candidates to install
usbip-core
usbip-vudc
vhci-hcd
dtoverlay=dwc2
dwc2
libcomposite
usb_f_rnidis
bash: git: command not found
bash: git: command not found
bash: cd: LD-ToyPad-Emulator: File or directory does not exist
sudo: curl: command not found
​

casioforever, 2024-01-26:
That's Why I Asked For An Offline Version Of This Emulator Here https://github.com/Berny23/LD-ToyPad-Emulator/issues/169. Apparently, The Developer Decided To Abandon His Project.

Berny23, 2024-01-26:
I did NOT abbandon my project. It is practically feature-complete and therefore a finished product that I don't want to ruin by constantly tinkering with it.
However, please feel free to create a pull request for your changes.

Since I switched from Windows to Linux completely in the meantime, I would need to set up everything for testing bigger changes, this could take a while and I'm happy to have other maintainers who help with GitHub issues. :)

casioforever, 2024-01-26:
No One Asked For A Change. And No, It's Not Feature-Complete. Constant Retrieval of your Files Through Internet, Makes Your Project Looks Like A Game Trainer That Requires Active Connection To Function Properly While The Game itself doesn't need one.

Luigimeansme, 2024-01-26:
> No One Asked For A Change. And No, It's Not Feature-Complete. Constant Retrieval of your Files Through Internet, Makes Your Project Looks Like A Game Trainer That Requires Active Connection To Function Properly While The Game itself doesn't need one. Btw. Your Code Doesn't Work On Debian v11.8.0.

Just because it requires an internet connection doesn't mean it's not complete. It functions perfectly to play the game, that's what it is built to do.

casioforever, 2024-01-26:
> Just because it requires an internet connection doesn't mean it's not complete. It functions perfectly to play the game, that's what it is built to do.
I Didn't See Any Emulator That Requires :
1 - Installing Linux Over Windows using VBox (Wait ! What ?)
2 - I Mean Specific Version Of Linux ( Working only on Debian 11.1.0)
3 - Provided Image Of Linux Requires Internet To Download GUI (Again ?)
4 - Requires Web Browser (Edge Or Chrome) + (Virtual Here ± Zadig) (WoooW)
Actually, This Emulator Misses A LOT...........................................................................................

Luigimeansme, 2024-01-26:
> I Didn't See Any Emulator That Requires : 1 - Installing Linux Over Windows using VBox (Wait ! What ?) 2 - I Mean Specific Version Of Linux (Working only on Debian 11.1.0) 3 - Provided Image Of [...]

You clearly don't understand what this is meant to do. This isn't meant to be a standalone program to run all of LEGO Dimensions, it's built to be useable on real hardware or via Cemu or RPCS3 to play Dimensions. The use of linux instead of Windows is because it's built using a raspberry pi, which uses Linux to run rather than Windows. Just because you don't understand it doesn't mean it's "missing a lot"

Berny23, 2024-01-26:
> I Didn't See Any Emulator That Requires :
> 1 - Installing Linux Over Windows using VBox (Wait ! What ?)
> 2 - I Mean Specific Version Of Linux (Working only on Debian 11.1.0)
> 3 - Provided Image Of [...]

Go somewhere else to annoy people.

You clearly have no idea how exactly an emulator like this works. This is not a normal program you install on a PC like Yuzu, Cemu, Citra etc.

And, as I already wrote, just add your own improvements in a pull request.

casioforever, 2024-01-26:
> And, as I already wrote, just add your own improvements in a pull request.

https://github.com/Berny23/LD-ToyPad-Emulator/issues/169

Luigimeansme, 2024-01-26:
> > And, as I already wrote, just add your own improvements in a pull request.
> 
> #169

That's not a pull request, you don't know how Github works. A pull request is editing the files yourself and submitting them to be added to the main project

casioforever, 2024-01-26:
>  which uses Linux to run rather than Windows. Just because you don't understand it doesn't mean it's "missing a lot"
You Should Visit A Doctor If You Can't See Clearly My Complaints
> (Working only on Debian 11.1.0)

casioforever, 2024-01-26:
> That's not a pull request, you don't know how Github works. A pull request is editing the files yourself and submitting them to be added to the main project
Just Say It Loud And Clear (I'm Too Lazy To Fix The Problems Of Non-Compatibility with Debian v11.8.0)

VladimirKuletski, 2024-01-26:
> Download:1 http://security.debian.org/debian-security bookworm-security InRelease [48.0 kB] Found:2 http://deb.debian.org/debian bookworm InRelease Found:3 http://deb.debian.org/debian [...]

Could you please try with `bullseye` image?

cort1237, 2024-01-26:
> I Didn't See Any Emulator That Requires :
>
> 1 - Installing Linux Over Windows using VBox (Wait ! What ?)
>
> 2 - I Mean Specific Version Of Linux (Working only on Debian 11.1.0)
>
> 3 - Provided Image Of [...]

That's because you don't understand what you're seeing. This is a repo for a program to turn a raspberry pi into an emulator for a toypad. It uses a web browser so it's headless and people who've never used a Pi don't have to buy and setup additional peripherals. Many people have played the entire game using this emulator.

The VM method is an alternative made later. It uses a VM in windows because the program need to run on a separate system so that the USB connection can be simulated. This makes it work with any popular emulator for the game.

I'm sorry if it's not working with your existing hardware. But as Berny said it's feature complete. When we make changes now, they tend to just be bug fixes or an occasional feature that would benefit everyone. If you really don't want to just pick up a raspberry pi, the entire repo is open to you. Make a branch and make it work for you. Just stop fussing about other people not doing that work for you. Drop the entitlement over a LEGO game.

casioforever, 2024-01-27:
> I'm sorry if it's not working with your existing hardware. But as Berny said it's feature complete
You Should Visit A Doctor If You Can't See Clearly My Complaints

> (Working only on Debian 11.1.0)

cort1237, 2024-01-27:
> You Should Visit A Doctor If You Can't See Clearly My Complaints
> 
> 
> 
> > (Working only on Debian 11.1.0)

*only tested on 11.1.0

I don't know what version is creating an incompatibility, because that was the only version we worked with at the time, three years ago. This is an old project that only gets occasional updates. Debian version compatibility was not a priority as, based on our interactions with users, most are using a VM or Pi dedicated to this program. So it didn't come up.

But I seriously doubt the incompatibility is anything that can't be overcome. If you stopped acting like a brat, and worked _with_ people I'm sure it could be solved. Instead you came in with a list of requests and when people didn't appear to fulfill them on demand, you decided to haunt our issue page for weeks and make constant insults to the people who have worked to build this program. Not cool.

casioforever, 2024-01-28:
> If you stopped acting like a brat, and worked with people I'm sure it could be solved. Instead you came in with a list of requests

Not before you stop acting like a thief, who steals the work of others and when somebody reports for a bug, you attacks him for not understanding how your stolen code works.

make your own code so you can understand it first. then complain about users who doesn't understand your effort. cursing testers isn't cool either.

cort1237, 2024-01-28:
There is no stolen code in this project, and I do have an understanding of it. I wrote most of it. These are again very rude accusations to fling at random people. You would do well to stop insulting the people on this repo. This project is worked on by good, passionate people who don't deserve this treatment from you. Now I'm going to be hiding this argument since it went far off topic to the original issue. I would appreciate if you stop this behavior. You are still free to use the issues page if you want to work through your problems in a collaborative manner. But any future off-topic posts or attacks on fellow users and collaborators will be deleted.

Ausbruhh, 2025-04-03:
> Could you please try with `bullseye` image?

hey i think im having this same problem, but i dont understand what "bullseye image" means, ive never touched vms or anything so i need a bit of guidance if its not too hard, thanks