Berny23/LD-ToyPad-Emulator #175: USBclient does not recognise Lego Reader 2.10

R4v4g3r1005, 2024-02-04:
I tried using this with Rpcs3 and used the troubleshooting method with Zhadig but the game kept shifting between 60 and 30 fps. I then tried using Cemu but it would not recognise the Toypad. I then went back to Rpcs3 but the USB client didn't recognise Lego Reader v2.10.
I even reinstalled Zhadig but it wouldn't recognise anything there either. Could anyone help with this please?

dverzolla, 2024-02-04:
Send logs etc.

R4v4g3r1005, 2024-02-04:
> […](#)

Forgive me, what logs?

R4v4g3r1005, 2024-02-06:
Forgive me, what are the logs? Where can I find them?

R4v4g3r1005, 2024-02-14:
> […](#)

WDYM? What logs?

R4v4g3r1005, 2024-02-17:
> […](#)
[vmware.log](https://github.com/Berny23/LD-ToyPad-Emulator/files/14318586/vmware.log)
[vmware-0.log](https://github.com/Berny23/LD-ToyPad-Emulator/files/14318587/vmware-0.log)
[vmware-1.log](https://github.com/Berny23/LD-ToyPad-Emulator/files/14318588/vmware-1.log)
[vmware-2.log](https://github.com/Berny23/LD-ToyPad-Emulator/files/14318589/vmware-2.log)

closed 2024-03-12 by R4v4g3r1005