Berny23/LD-ToyPad-Emulator #182: Unable to get past "npm install" step

derpytails, 2024-03-26:
I'm trying to follow the guide as close as possible and I get this for whenever I do "npm install"

> @serialport/bindings@9.2.8 install /home/cybertayls/LD-ToyPad-Emulator/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install warn install No prebuilt binaries found (target=11.15.0 runtime=node arch=x64 libc= platform=linux)
/home/cybertayls/.nvm/versions/node/v11.15.0/lib/node_modules/node-gyp/bin/node-gyp.js:81
    const args = await prog.commands[command.name](command.args) ?? []
                                                                  ^

SyntaxError: Unexpected token ?
    at Module._compile (internal/modules/cjs/loader.js:760:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.2.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@9.2.8 install script.

If it helps. I'm using VMWare Workstation Player

LizDerby140, 2025-05-12:
I'm having the same issue and I dont know how to resolve it, please help

JonasBrinkmann-DE, 2025-05-12:
Try running: 
npm install --global node-gyp@8.4.1
npm config set node_gyp $(npm prefix -g)/lib/node_modules/node-gyp/bin/node-gyp.js