Berny23/LD-ToyPad-Emulator #184: The pi zero 2 w seems to not be able to connect through usb to a PlayStation 5

ignacioat2, 2024-04-08:
Is there a reason this might not work with a raspberry pi zero 2 W. I did the whole process right, but when I connect it to my PlayStation 5 it says “connect the Toypad” even though the pi is already connected with a micro usb with data transfer from the usb port of the pi. Maybe any solutions to this?

ignacioat2, 2024-04-08:
I was able to do the complete installation and got it to work with a normal pi zero w but im trying with the pi zero 2 w and i dont know if they use the same cable for data transfer. For example the pi zero w uses the one that the ps4 controllers use micro usb b i think but i dont know if the pi zero 2w uses the same or not, that might be the problem.

VladimirKuletski, 2024-04-09:
Pi Zero 2W doesn't have port which supports OTG mode. So it'w Pi Zero W only. 

Pi 4 has an OTG feature, but hasn't been checked yet by this community.

ignacioat2, 2024-04-09:
So there is no way that i can use the pi zero 2 w for data transfer with the usb? Even though the pi zero 2w is supposed to be just a more powerful pi zero 2w. Please if you know how to do it I will appreciate it a lot. Thanks either way

ignacioat2, 2024-04-09:
i got it working i dont know how

ignacioat2, 2024-04-09:
I got it working once after using some code from this page https://forums.raspberrypi.com/viewtopic.php?t=360603#p2163877 but I cant quite figure out which line is necessary for the pi to work and also how to make it like always work the OTG without writing some lines everytime.

VladimirKuletski, 2024-04-10:
Don't have pi zero 2 in my collection. So, won't be able to check.

Florin9doi, 2024-04-30:
RPi Zero 2W is compatible, but the recent RPI OS releases have changed the `config.txt` path to `/boot/firmware/config.txt`.
You'll have to update the step #<span>3 from [Installation](https://github.com/Berny23/LD-ToyPad-Emulator?tab=readme-ov-file#option-2-single-board-computer):
```diff
- echo "dtoverlay=dwc2" | sudo tee -a /boot/config.txt
+ echo "dtoverlay=dwc2" | sudo tee -a /boot/firmware/config.txt
```
or 
```diff
- echo "dtoverlay=dwc2" | sudo tee -a /boot/config.txt
+ [[ -f /boot/firmware/config.txt ]] && ( echo "dtoverlay=dwc2" | sudo tee -a /boot/firmware/config.txt ) || ( echo "dtoverlay=dwc2" | sudo tee -a /boot/config.txt )
```

PR #205 (2024-08-05): Update config.txt path
Closes #184, Closes #189
closed 2024-08-05 by Berny23