Berny23/LD-ToyPad-Emulator #185: Package 'python' has no installation candidate?

Sonkinator1991, 2024-04-09:
So, I've been trying to get this to work for a while, but that first line of code I have to paste in terminal just throws me a Package 'python' has no installation candidate. I'm not sure how to fix this, can you help me?

DiverseBloom780, 2024-08-04:
> So, I've been trying to get this to work for a while, but that first line of code I have to paste in terminal just throws me a Package 'python' has no installation candidate. I'm not sure how to fix this, can you help me?

Is it with a Virtual Machine? cause i had this issue with the VM and then i found a fix

Sonkinator1991, 2024-08-04:
yea im on vmware

clintonb, 2024-08-19:
Python 2.x is provided by the `python` and reached end of life years ago. You will need `python3`; however, the Python code being run (and written for Python 2) may be incompatible with Python 3.