Berny23/LD-ToyPad-Emulator #192: Raspberry pi zero 2w issue

SpaceJamp, 2024-06-04:
mech@raspberrypi:~/LD-ToyPad-Emulator $ node index.js
fs.js:119
    throw err;
    ^

Error: ENOENT: no such file or directory, open '/dev/hidg0'
    at Object.openSync (fs.js:448:3)
    at new RawTransport (/home/mech/LD-ToyPad-Emulator/node_modules/node-ld/dist/transports/raw.js:11:27)
    at new ToyPadEmu (/home/mech/LD-ToyPad-Emulator/node_modules/node-ld/dist/lib/ToyPadEmu.js:65:69)
    at Object.<anonymous> (/home/mech/LD-ToyPad-Emulator/index.js:24:10)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11

KieronTheKiwi, 2024-06-04:
Same issue. :(

pianistrevor, 2024-07-11:
@SpaceJamp @KieronTheKiwi The location of config.txt has changed.
- Go to /boot/config.txt and delete the line "dtoverlay=dwc2".
- Go to /boot/firmware/config.txt and add the line "dtoverlay=dwc2".
- Reboot your device

Worked for me.

DiverseBloom780, 2024-08-05:
> @SpaceJamp @KieronTheKiwi The location of config.txt has changed.
> 
> * Go to /boot/config.txt and delete the line "dtoverlay=dwc2".
> * Go to /boot/firmware/config.txt and add the line "dtoverlay=dwc2".
> * Reboot your device
> 
> Worked for me.

where at in the firmware/config.txt do we add dtoverlay=dwc2?

closed 2024-08-05 by SpaceJamp

DiverseBloom780, 2024-08-05:
> @SpaceJamp @KieronTheKiwi The location of config.txt has changed.
> 
> * Go to /boot/config.txt and delete the line "dtoverlay=dwc2".
> * Go to /boot/firmware/config.txt and add the line "dtoverlay=dwc2".
> * Reboot your device
> 
> Worked for me.

could you give me a pic so i can see where to add and which to delete? in my  /boot/config.txt there is multiple entries, not sure where to add dtoverlay=dwc2 in the /boot/firmware/config.txt

trevor-natiuk, 2024-08-05:
> could you give me a pic so i can see where to add and which to delete? in my  /boot/config.txt there is multiple entries, not sure where to add dtoverlay=dwc2 in the /boot/firmware/config.txt

What does yours have? According to the setup script in the README it should just be the one line. Anything else would come from raspi-config.