Berny23/LD-ToyPad-Emulator #194: Lego Dimensions won’t connect to Wii u

KieronTheKiwi, 2024-06-15:
I have a Raspberry Pi Zero W with Raspberry Pi OS Lite (legacy, bullseye) installed (because that was the only one I could get to work) and it won’t communicate with the Wii U, I have the server up and running but no connection, any suggestions?

closed 2024-06-17 by KieronTheKiwi

VladimirKuletski, 2024-06-17:
Please give this: https://github.com/Berny23/LD-ToyPad-Emulator/discussions/188 a try.
Should be easier to start with.

KieronTheKiwi, 2024-06-17:
I fixed it by using Raspberry Pi Lite Legacy Bullseye instead of Raspberry Pi Lite bookworm otherwise you will get error such file or directory, open '/dev/hidg0' and then tried a bunch of different cables in order to ensure it has data transmission but thank you :)

VladimirKuletski, 2024-06-17:
Also had issues with the latest releases. Will  update the "step-by-step" guide.