Berny23/LD-ToyPad-Emulator #198: Can you open the emulator page on a mobile device?

Zephereth, 2024-06-26:
I got everything working perfectly and easily, really appreciate the hard work on this project and the simple instructions!  I was just wondering if it's possible to open debian/ in the mobile browser to make it easier to swap things around, otherwise I need to alt-tab each time since I only have one monitor.

irrdkwhattoput, 2024-06-26:
Yes, you can enter the page via the machine IPV4

Zephereth, 2024-06-26:
> Yes, you can enter the page via the hosting machine IPV4

I've tried that and it doesn't work for me. :(  Is there something I have to do prior for it to work?  I read something about being on the same network, etc., which my PC is on LAN, phone is Wi-Fi, of course.

irrdkwhattoput, 2024-06-26:
> > Yes, you can enter the page via the hosting machine IPV4
> 
> I've tried that and it doesn't work for me. :(  Is there something I have to do prior for it to work?  I read something about being on the same network, etc., which my PC is on LAN, phone is Wi-Fi, of course.

Do you have bridged network enabled in your VM? you need to be on the same network as the hosting device to access the page

Zephereth, 2024-06-26:
> Do you have bridged network enabled in your VM? you need to be on the same network as the hosting device to access the page

I think I tried yesterday with the bridged network option enabled, do I need to check the "replicate physical connection state" checkbox as well?

Zephereth, 2024-06-26:
I'm at work atm and have been playing around with it on my work PC, but the way the network is setup might be disallowing it or something.  Using the bridge network option doesn't allow the VM to connect to the internet, I can't remember about at home though.

Zephereth, 2024-06-26:
> Do you have bridged network enabled in your VM? you need to be on the same network as the hosting device to access the page

Tried it again at home with bridged enabled, also disabled the firewall for a minute while I checked to allow traffic if not already and it still refuses to connect.  My phone is on Wi-Fi, which I'm assuming is why it's not working?  But I wouldn't be able to use bridged if I changed the PC to Wi-Fi as well, so any suggestions?