Berny23/LD-ToyPad-Emulator #201: Can't run on PS3

Bandud2, 2024-07-30:
For some reason I can't seem to get the emulator running when connected to my PS3. I ran the command to enable the automatic start up, and it works when connected to my computer, as I was able to connect it to my laptop, and was able to access the emulator from the browser without doing anything. When I connect it to my PS3 however, my raspberry pi does work ( It emits the green led) but I can't access the emulator from my browser. And yes, I am using a micro usb cable.

Bandud2, 2024-07-30:
> What was solved: running pi or emulator (when connected to USB)?
> 
> Please give this: #188 a try. Should be easier to troubleshoot.

Both, the pi is running and the emulator works (I can connect to it with the browser). But it's not working ingame and I am getting the message to connect the toy pad to the console

VladimirKuletski, 2024-07-30:
> Both, the pi is running and the emulator works (I can connect to it with the browser). But it's not working ingame and I am getting the message to connect the toy pad to the console

Please doublecheck the 3rd item from the guide - this is the most common issue. Either skip it or take from VM chapter

Bandud2, 2024-07-30:
> Please doublecheck the 3rd item from the guide - this is the most common issue. Either skip it or take from VM chapter

Do you mean the configuration step?

Bandud2, 2024-07-31:
> What was solved: running pi or emulator (when connected to USB)?
> 
> Please give this: #188 a try. Should be easier to troubleshoot.

Ok so not only did this method not work, but now after I tried to install the thing like I did before, the node index.js command gives an error
[image]

Bandud2, 2024-07-31:
So after trying different fixes, I'm pretty much back at square 1. The emulator does start automatically when connected to my PS3, but the actual game doesn't detect it.

Bandud2, 2024-08-01:
@Berny23 Do you think you can help me out with this? I'm honestly clueless as to why it isn't working

Berny23, 2024-08-01:
usb_setup_script.sh creates the virtual device for USB communication. Your error code shows that the script didn't run properly.

Bandud2, 2024-08-01:
> usb_setup_script.sh creates the virtual device for USB communication. Your error code shows that the script didn't run properly.

If you're talking about the ENOENT error then I already resolved it. I managed to get the emulator running on the pi, it just doesn't work in the game.

Bandud2, 2024-08-01:
Ok nvm, I managed to fix my issue and it was honestly such a stupid one... Basically I was using  cable that, was capable of connecting the pi to the ps3 for the pi to be detected as a USB device, it didn't have enough power to do so.. Then I had another cable which was, funnily enough the exact opposite. It could supply the Pi with the power to actually run the emulator, but for some reason when I use it, my Pi is never detected. What I did was to connect the latter cable to the edge port of the pi, just for it to boot up and run the emulator, then connect the first cable to the "main" port so the game would actually recognize it. It's funny that if only I had a normal cable that had both capabilities, I wouldn't even have this issue...

closed 2024-08-01 by Bandud2