Berny23/LD-ToyPad-Emulator #202: Can't get PS4 to recognize Pi Zero or get node to run

DiverseBloom780, 2024-08-03:
I'm needing someone to help cause I can't figure this out here, I followed the guide and I can't get the node to work,  and come to find out after running ls command under the Toypad folder that the node is not even there, am I doing something wrong here? Or is there modules or something different I'm supposed to do here? Been up two nights in a row here trying to figure this out

Benderowy, 2024-08-04:
https://github.com/Berny23/LD-ToyPad-Emulator/issues/192#issuecomment-2221886379

DiverseBloom780, 2024-08-05:
> [#192 (comment)](https://github.com/Berny23/LD-ToyPad-Emulator/issues/192#issuecomment-2221886379)

thanks i will try it out and see where it goes from there, hopefully it fixes the issue for me

DiverseBloom780, 2024-08-05:
> [#192 (comment)](https://github.com/Berny23/LD-ToyPad-Emulator/issues/192#issuecomment-2221886379)

i got it to read as Lego Reader V2.10

# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
# Enable USB OTG for all Raspberry Pi models
dtoverlay=dwc2

 this is what fixed it

closed 2024-08-05 by DiverseBloom780