Berny23/LD-ToyPad-Emulator #227: Is it possible to use Orange pi zero 2w instead of Raspberry Pi?

Bibasyoy, 2024-12-25:
Is it possible to use Orange Pi zero 2w instead of Raspberry pi? It also should support Raspberry OS, but I have never had single board computers before and I am new to this topic. Thx.

DomanDragon, 2025-03-16:
According to a quick search, it supports USB Gadget, which is the main requirement for this to work. So yes, it's possible.

Erich112, 2025-04-12:
Hello,
I am currently trying this on such board and I have managed to get the server part running (I did run into some errors, it seems like its not communicating with the game at all as i just have the "please connect the toy pad to the console" message, I'm playing on real Wii U)

One thing that wasn't mentioned is that I kept running into an error when installing it initially
```
gyp ERR! build error
gyp ERR! stack Error: not found: make
```
and I found out I had to run an extra command
`sudo apt-get install build-essential`.
I'll let you know if I do manage to get it working with the game