Berny23/LD-ToyPad-Emulator #231: Error: listen EACCES: permission denied 0.0.0.0:80

jonbon1912, 2025-01-25:
Hello can anyone help me fix this error? I'm using raspberry pi 4 and when I updated node, it gives me the following error:

node:events:502
      throw er; // Unhandled 'error' event
      ^

Error: listen EACCES: permission denied 0.0.0.0:80
    at Server.setupListenHandle [as _listen2] (node:net:1915:21)
    at listenInCluster (node:net:1994:12)
    at Server.listen (node:net:2099:7)
    at Object.<anonymous> (/home/jonpi/LD-ToyPad-Emulator/index.js:616:8)
    at Module._compile (node:internal/modules/cjs/loader:1562:14)
    at Object..js (node:internal/modules/cjs/loader:1699:10)
    at Module.load (node:internal/modules/cjs/loader:1313:32)
    at Function._load (node:internal/modules/cjs/loader:1123:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:217:24)
Emitted 'error' event on Server instance at:
    at emitErrorNT (node:net:1973:8)
    at process.processTicksAndRejections (node:internal/process/task_queues:90:21) {
  code: 'EACCES',
  errno: -13,
  syscall: 'listen',
  address: '0.0.0.0',
  port: 80
}

A fix will be much appreciated thanks.

VladimirKuletski, 2025-01-26:
It might be related to port number and unprivileged user.
Please review: https://stackoverflow.com/questions/60372618/nodejs-listen-eacces-permission-denied-0-0-0-080

jonbon1912, 2025-01-26:
Oh now I have this error:

Error: listen EADDRINUSE: address already in use :::80